zephyrproject-rtos/zephyr #33381: West debug does not work with Bluetooth shell and nRF52840 DK

mkarhumaa, 2021-03-16:
**Describe the bug**
`west debug` does not work anymore with Bluetooth shell on nRF52840 DK. According to git bisect, commit [enabling null pointer detection in tests](https://github.com/zephyrproject-rtos/zephyr/commit/1b22f6b8c8531c06d4d644b8bf18b729ecdc7db4) caused this problem.

Inspired by [this change](https://github.com/zephyrproject-rtos/zephyr/pull/32983) I was able to figure out workaround that fixes the problem:
```
diff --git a/boards/arm/nrf52840dk_nrf52840/Kconfig.defconfig b/boards/arm/nrf52840dk_nrf52840/Kconfig.defconfig
index bc06b9074f..0192cd9217 100644
--- a/boards/arm/nrf52840dk_nrf52840/Kconfig.defconfig
+++ b/boards/arm/nrf52840dk_nrf52840/Kconfig.defconfig
@@ -21,4 +21,8 @@ endif # USB
 config BT_CTLR
        default BT
+choice CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION
+       default CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_NONE
+endchoice
+
 endif # BOARD_NRF52840DK_NRF52840
```

**To Reproduce**
Steps to reproduce the behavior:
1. `cd tests/bluetooth/shell`
2. `west build -b nrf52840dk_nrf52840`
3. open console for shell: `sudo minicom -D /dev/ttyACM0 -b 115200`
4. `west debug`
5. In gdb prompt type `continue` to try to start running the program
6. Interactive shell does not start 

**Expected behavior**
When typing "continue" in gdb test application, Bluetooth shell in this case, should start running.

**Impact**
annoyance, workaround exists

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: master (4aed856d7fb17ced2554aaebe5a3c3069d127a74)

carlescufi, 2021-03-16:
I am hitting this as well, and I would say this is at least medium prio, it's breaking the Bluetooth shell for most of us.

nashif, 2021-05-25:
close as duplicate to #34222

closed 2021-05-25 by nashif