zephyrproject-rtos/zephyr #33403: trigger_fault_divide_zero test case didn't run divide instruction

IRISZZW, 2021-03-17:
**Describe the bug**
the test case of trigger_fault_divide_zero not really test divide zero function, indeed the compiler do some optimization for it,
just trigger exception.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_cortex_m3 tests/ztest/error_hook/
2. west build -b nsim_em tests/ztest/error_hook/
3. objdump -S build/CMakeFiles/app.dir/src/main.c.obj

**source codes**
https://github.com/zephyrproject-rtos/zephyr/blob/35f0ed23c9b6980bed20ed90aef99cdb290e7e30/tests/ztest/error_hook/src/main.c#L79-L87

https://github.com/zephyrproject-rtos/zephyr/blob/35f0ed23c9b6980bed20ed90aef99cdb290e7e30/tests/ztest/error_hook/src/main.c#L158-L194

**Disassembly codes of qemu_cortex_m3**
```
  50:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  54:	deff      	udf	#255	; 0xff     ##no divide instruction here
  56:	2001      	movs	r0, #1
```
```
Disassembly of section .text.tThread_entry: 

00000000 <tThread_entry>:
   0:	4b1f      	ldr	r3, [pc, #124]	; (80 <tThread_entry+0x80>)
   2:	b513      	push	{r0, r1, r4, lr}
   4:	6804      	ldr	r4, [r0, #0]
   6:	6819      	ldr	r1, [r3, #0]
   8:	481e      	ldr	r0, [pc, #120]	; (84 <tThread_entry+0x84>)
   a:	f7ff fffe 	bl	0 <printk>
   e:	2000      	movs	r0, #0
  10:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  14:	2c04      	cmp	r4, #4
  16:	d80f      	bhi.n	38 <tThread_entry+0x38>
  18:	e8df f004 	tbb	[pc, r4]
  1c:	1d191303 	.word	0x1d191303
  20:	27          	.byte	0x27
  21:	00          	.byte	0x00
  22:	2001      	movs	r0, #1
  24:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  28:	4b17      	ldr	r3, [pc, #92]	; (88 <tThread_entry+0x88>)
  2a:	4818      	ldr	r0, [pc, #96]	; (8c <tThread_entry+0x8c>)
  2c:	f8d3 3fff 	ldr.w	r3, [r3, #4095]	; 0xfff
  30:	9301      	str	r3, [sp, #4]
  32:	9901      	ldr	r1, [sp, #4]
  34:	f7ff fffe 	bl	0 <printk>
  38:	b002      	add	sp, #8
  3a:	e8bd 4010 	ldmia.w	sp!, {r4, lr}
  3e:	f7ff bffe 	b.w	0 <ztest_test_fail>
  42:	2001      	movs	r0, #1
  44:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  48:	ab01      	add	r3, sp, #4
  4a:	4798      	blx	r3
  4c:	e7f4      	b.n	38 <tThread_entry+0x38>
  4e:	2001      	movs	r0, #1
  50:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  54:	deff      	udf	#255	; 0xff     ##no divide instruction here
  56:	2001      	movs	r0, #1
  58:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  5c:	4040      	eors	r0, r0
  5e:	f380 8811 	msr	BASEPRI, r0
  62:	f04f 0004 	mov.w	r0, #4
  66:	df02      	svc	2
  68:	e7e6      	b.n	38 <tThread_entry+0x38>
  6a:	2001      	movs	r0, #1
  6c:	f7ff fffe 	bl	0 <z_impl_ztest_set_fault_valid>
  70:	4040      	eors	r0, r0
  72:	f380 8811 	msr	BASEPRI, r0
  76:	f04f 0003 	mov.w	r0, #3
  7a:	df02      	svc	2
  7c:	e7dc      	b.n	38 <tThread_entry+0x38>
  7e:	bf00      	nop
	...
  88:	fffff000 	.word	0xfffff000
  8c:	00000011 	.word	0x00000011
```
**Disassembly codes of nsim_em**

```
		ztest_set_fault_valid(true);
  68:	710c                	mov_s	r0,1
  6a:	0802 0000           	bl	0	;0 <tThread_entry>
	a = a / b;
  6e:	78be                	trap_s	0x5  ##no divide instruction here
```

```
Disassembly of section .text.tThread_entry:

00000000 <tThread_entry>:
{
   0:	c2e2                	enter_s	[r13,blink]
   2:	c1a1                	sub_s	sp,sp,0x4
	printk("case type is %d\n", case_type);
   4:	1600 7001 0000 0000 	ld	r1,[0]
	int sub_type = *(int *)p1;
   c:	80a0                	ld_s	r13,[r0,0]
	printk("case type is %d\n", case_type);
   e:	40c3 0000 0000      	mov_s	r0,0
  14:	0802 0000           	bl	0	;0 <printk>
	ztest_set_fault_valid(false);
  18:	700c                	mov_s	r0,0
  1a:	0802 0000           	bl	0	;0 <tThread_entry>
	switch (sub_type) {
  1e:	0d39 1155           	brhs.nt	r13,0x5,56	;54 <tThread_entry+0x54>
  22:	2632 7340 0000 0000 	ldb	r0,[0,r13]
  2a:	70c3 0000 0000      	add_s	r0,r0,0
  30:	7800                	j_s	[r0]
  32:	2a06 3e36           	adds	r54,gp,r56
  36:	004a 710c           	ble	925770	;e207e <tThread_entry+0xe207e>
		ztest_set_fault_valid(true);
  3a:	0802 0000           	bl	0	;0 <tThread_entry>
	volatile int b = *((int *)a);
  3e:	1600 7000 ffff ffff 	ld	r0,[0xffffffff]
  46:	c040                	st_s	r0,[sp,0]
	printk("b is %d\n", b);
  48:	4187                	ld_s	r1,[sp,0]
  4a:	40c3 0000 0000      	mov_s	r0,0
  50:	0802 0000           	bl	0	;0 <printk>
}
  54:	7487                	add_s	sp,sp,4
  56:	c2c2                	leave_s	[r13,blink]
	ztest_test_fail();
  58:	0001 0000           	b	0	;0 <ztest_test_fail>
		ztest_set_fault_valid(true);
  5c:	710c                	mov_s	r0,1
  5e:	0802 0000           	bl	0	;0 <tThread_entry>
	((void(*)(void))&a)();
  62:	4083                	mov_s	r0,sp
  64:	7840                	jl_s	[r0]
}
  66:	f1f8                	b_s	-16	;54 <tThread_entry+0x54>
		ztest_set_fault_valid(true);
  68:	710c                	mov_s	r0,1
  6a:	0802 0000           	bl	0	;0 <tThread_entry>
	a = a / b;
  6e:	78be                	trap_s	0x5  ##no divide instruction here
		ztest_set_fault_valid(true);
  70:	710c                	mov_s	r0,1
  72:	0802 0000           	bl	0	;0 <tThread_entry>
	k_panic();
  76:	204a 0100           	mov	r0,0x4
  7a:	785e                	trap_s	0x2
		ztest_set_fault_valid(true);
  7c:	710c                	mov_s	r0,1
  7e:	0802 0000           	bl	0	;0 <tThread_entry>
	k_oops();
  82:	204a 00c0           	mov	r0,0x3
  86:	785e                	trap_s	0x2
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit (Master branch)

enjiamai, 2021-03-17:
@IRISZZW  Got it, thanks for your analysis, will take a look at this as soon.

ioannisg, 2021-03-17:
> @IRISZZW Got it, thanks for your analysis, will take a look at this as soon.

We probably need to do a -O0 around the divide-by-zero code; I've done similar things in other tests.

ioannisg, 2021-03-17:
Adding LOW priority; does not seem to be a (very) serious issue.

enjiamai, 2021-03-17:
> >  We probably need to do a -O0 around the divide-by-zero code; I've done similar things in other tests.

Good suggestion,  thank you for sharing this information!

ioannisg, 2021-03-17:
> > > We probably need to do a -O0 around the divide-by-zero code; I've done similar things in other tests.
> 
> Good suggestion, thank you for sharing this information!

Will you try to provide a fix for it, @enjiamai ?

ioannisg, 2021-03-17:
> @ioannisg Excuse me, if you have a solution and prefer doing this, please go-ahead to fix it. My plan is I should fix it, but I might do it after tomorrow. How do you think? Thank you!

Please, go ahead with sending a fix, when you have time!

enjiamai, 2021-03-17:
> Please, go ahead with sending a fix, when you have time!

Thank you!! ^^

IRISZZW, 2021-03-18:
> Adding LOW priority; does not seem to be a (very) serious issue.
@ioannisg @enjiamai 

These may not be a serious issue, but we must pay attention for it, compiler may do some aggressive optimizations to  cheat us, 
especially in some test cases. we may pass the test cases, but do those test case really test the function we want to test?

if you check the disassembly codes of 
```
 trigger_fault_access(); 
 trigger_fault_illeagl_instuction(); 
 trigger_fault_divide_zero(); 
```
all have been optimized by compiler, I believe there must be some similar optimizations in other tests.

enjiamai, 2021-03-18:
> all have been optimized by compiler, I believe there must be some similar optimizations in other tests.
@IRISZZW Yes, totally agree with you, many thanks for pointing this out.

PR #33886 (2021-05-04): tests: ztest: do not optimize the divide by zero test case
Make the divide instruction not being optimized of the ztest example
test case. Do this because it will be replaced by invalid opcode
exception instruction generated by GCC.

This also prevents optimizing from 1. access the illegal address. 2. run the illegal instruction.
Because GCC will also optimize them in this case.

Fixes #33403.
commit 86851434 (2021-05-04): tests: ztest: do not optimize the divide by zero test case

Make the divide instruction not being optimized of the ztest example
test case. Do this because it will be replaced by invalid opcode
exception instruction generated by GCC.

Fixes #33403.
closed 2021-05-04 by nashif