zephyrproject-rtos/zephyr #33422: samples/subsys/usb/dfu/sample.usb.dfu fails on multiple platforms in daily build

galak, 2021-03-17:
`./scripts/twister -s samples/subsys/usb/dfu/sample.usb.dfu  -l`

gets:

```
ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for nucleo_f207zg

ERROR   - nucleo_f207zg             samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nucleo_f207zg/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   33/  67  49%  skipped:  297, failed:    1ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1060_evk

ERROR   - mimxrt1060_evk            samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1060_evk/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   34/  67  50%  skipped:  297, failed:    2ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1010_evk

ERROR   - mimxrt1010_evk            samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1010_evk/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   35/  67  52%  skipped:  297, failed:    3ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1020_evk

ERROR   - mimxrt1020_evk            samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1020_evk/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   36/  67  53%  skipped:  297, failed:    4ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1060_evk_hyperflash

ERROR   - mimxrt1060_evk_hyperflash samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1060_evk_hyperflash/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   37/  67  55%  skipped:  297, failed:    5ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1015_evk

ERROR   - mimxrt1015_evk            samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1015_evk/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   38/  67  56%  skipped:  297, failed:    6ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1050_evk

ERROR   - mimxrt1050_evk            samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1050_evk/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   39/  67  58%  skipped:  297, failed:    7ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/subsys/usb/dfu for mimxrt1050_evk_qspi

ERROR   - mimxrt1050_evk_qspi       samples/subsys/usb/dfu/sample.usb.dfu              FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1050_evk_qspi/samples/subsys/usb/dfu/sample.usb.dfu/build.log
INFO    - Total complete:   67/  67  100%  skipped:  324, failed:    8
INFO    - 0 of 8 test configurations passed (0.00%), 8 failed, 324 skipped with 0 warnings in 13.70 seconds
```

This us due to this commit: 896c29f494ac63df8454daa40fd40f2f7b1d1601

jfischer-no, 2021-03-17:
The reason for 896c29f was to filter out these platforms, I really have no idea why it does not work.

MaureenHelm, 2021-03-19:
The only i.MX RT board that defines fixed-partitions is `mimxrt1064_evk`. None of the others define it, so it looks like the filter isn't working.

jfischer-no, 2021-03-19:
How I understood, the problem is that this boards fail during "Loaded configuration/Merged configuration" before filtering.

```
warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) has direct dependencies FLASH_HAS_PAGE_LAYOUT && FLASH with value n, but is currently being y-selected by the following symbols:
 - STREAM_FLASH (defined at subsys/storage/stream/Kconfig:7), with value y, direct dependencies y (value: y)

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) has direct dependencies FLASH_HAS_DRIVER_ENABLED with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_IMG_MANAGER (defined at subsys/dfu/Kconfig:25), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)
```

galak, 2021-03-24:
@jfischer-no any updates on this?

jfischer-no, 2021-03-24:
> @jfischer-no any updates on this?

Well I see this more as a problem of the corresponding boards/platforms.
I think reverting 896c29f or to exclude affected platforms in the sample does not move us forward.

de-nordic, 2021-03-29:
Maybe split the process in to two phases where the first one would be to generate all the files needed (like in case of build --cmake-only) and the second would be actual build and run.
The filters would have to be checked between phases.

PR #33879 (2021-03-31): samples: usb: dfu: exclude platforms sample can't be built on
Several platforms don't have flash drivers and these platforms the DFU
sample will not build or even pass Kconfig on.  Exclude them until they
have flash driver support.

Fixes #33422
commit a5ea06d5 (2021-03-31): samples: usb: dfu: exclude platforms sample can't be built on

Several platforms don't have flash drivers and these platforms the DFU
sample will not build or even pass Kconfig on.  Exclude them until they
have flash driver support.

Fixes #33422
closed 2021-03-31 by MaureenHelm