zephyrproject-rtos/zephyr #33449: Remove deprecated items in 2.7

galak, 2021-03-17:
Here's a list we should update on things deprecated in 2.5 that should be removed in zephyr 2.7

- [ ] STM32 pinmux macros
- [ ] bt_set_id_addr - include/bluetooth/bluetooth.h
- [ ] DEVICE_INIT / DEVICE_AND_API_INIT - include/device.h
- [ ] BUILD_ASSERT_MSG - include/toolchain/common.h, include/toolchain/gcc.h
- [ ] GET_ARG2 / GET_ARGS_LESS_1 - include/sys/util_macro.h
- [ ] CONFIG_FS_LITTLEFS_FC_MEM_POOL - subsys/fs/Kconfig.littlefs 
- [ ] CONFIG_PRINTK64 in lib/os/Kconfig

- [ ] CONFIG_MULTITHREADING - kernel/Kconfig (not 100% sure on this one) ???

nashif, 2021-03-17:
>     * CONFIG_MULTITHREADING - kernel/Kconfig (not 100% sure on this one) ???

we need to drop this or fully support it and overhaul the code before 2.6 is released. Actually this should have happened in 2.5, so, this should be done earlier than 2.7

nashif, 2021-03-17:
https://github.com/zephyrproject-rtos/zephyr/issues/27415#issuecomment-685833663

galak, 2021-03-18:
> we need to drop this or fully support it and overhaul the code before 2.6 is released. Actually this should have happened in 2.5, so, this should be done earlier than 2.7
> 
> [#27415 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/27415#issuecomment-685833663)

yeah, wasn't sure what @carlescufi is looking to do in this area for 2.6.

carlescufi, 2021-03-23:
> yeah, wasn't sure what @carlescufi is looking to do in this area for 2.6.

I am looking into this for the 2.6 timeline.

galak, 2021-03-24:
```
arch/arm/include/aarch32/cortex_r/exc.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/aarch32/cortex_r/stack.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_m/cmse.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_m/exc.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_m/stack.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_m/tz.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_r/exc.h:#ifndef CONFIG_COMPAT_INCLUDES
arch/arm/include/cortex_r/stack.h:#ifndef CONFIG_COMPAT_INCLUDES
include/arch/arm/aarch32/cortex_r/cmsis.h:#ifndef CONFIG_COMPAT_INCLUDES
include/arch/arm/aarch32/cortex_r/cpu.h:#ifndef CONFIG_COMPAT_INCLUDES
include/arch/arm/aarch32/cortex_r/sys_io.h:#ifndef CONFIG_COMPAT_INCLUDES
```

galak, 2021-06-10:
https://github.com/zephyrproject-rtos/zephyr/pull/36134

galak, 2021-06-15:
@de-nordic can you possibly look at the cleanup/removal of deprecated code in the littlefs code?

de-nordic, 2021-06-16:
> 
> 
> @de-nordic can you possibly look at the cleanup/removal of deprecated code in the littlefs code?

Here: https://github.com/zephyrproject-rtos/zephyr/pull/36285

cfriedt, 2021-07-09:
@galak - looks as though this one is done - are there any other deprecated items to remove?

mniestroj, 2021-07-10:
Deprecated DT properties: #36862 #36863

closed 2021-07-12 by galak