zephyrproject-rtos/zephyr #33483: TIMESLICE and PM interaction and expected behavior

ceolin, 2021-03-18:
**Describe the bug**

When `CONFIG_TIMESLICE` is set and different from 0, the first time the idle thread runs  `pm_system_suspend` is called with the programmed slice. The system does not renew the slice  if the only thread running is the idle, so next time `pm_system_suspend` is called with remaining time to the scheduled event less the slice time that was evaluated. Wrapping up, if an application calls `k_msleep(5000)` and `TIMESLICE` is set to 2000 (for example), `pm_system_suspend` will be called with ~2000 the first time and then with ~3000. In this case, 3000 may not be enough to the system go to the power state that the application was expecting.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable `CONFIG_TIMESLICE`
2. Set `CONFIG_TIMESLICE_SIZE` different from 0
3. Put the application to sleep with some value greater than the `CONFIG_TIMESLICE_SIZE`

**Expected behavior**
???? This seems to be right, but the documentation should reflect that.

**Impact**
Applications using TIME SLICE should understand the system behavior when making assumptions about  power management.

ceolin, 2021-03-18:
@nashif  @erwango @gmarull

galak, 2021-05-19:
@ceolin is this still an issue?

andyross, 2021-05-21:
This sounds like a bug to me.  The scheduler understands that the idle thread is not "sliceable()", but the implication is that on entry to the idle thread the existing timeout is left in place and not cancelled.  It's true that spurious timer interrupts are not normally considered a bug, but they really should be in idle (for exactly this reason).

andyross, 2021-05-21:
It's a scheduler bug, not a PM thing, basically.  I'll take it.  As far as release planning goes: not sure this rises to "medium" personally, as the impact is a single wakeup event per idle.  That's important for PM tuning only on very power-constrained apps.  But that's someone else's call.

ceolin, 2021-05-21:
Just to be clear, this happens just the first time the idle thread runs. So the idle thread runs with the next timeout being the slice and after that if there is not active the slice is not renewed and the idle thread gets the remaining time to the scheduled event. It is not the best thing but it is not major bug IMHO. Actually, this could even be documented as part of the system behavior.

andyross, 2021-05-21:
OK, got it.  Yeah, actually looking at the code it's doing the right thing and only resetting the timeout when the new thread is not the idle thread.

A different way to explain this is that it's not the "idle thread's" timeout that's expiring, it's whatever was running earlier.  If you have a timesliced thread that starts running and gets a timeout set for 100 ticks in the future, but then it immediately goes to sleep, then the idle thread will run but the **old** timeout is still active and will still expire.

The system doesn't have the ability to "cancel" a timeslice expiration right now (for historical reasons having to do with the way we've always multiplexed the timer interrupt).  It's been a nice-to-have for a while (I think there's an enhancement filed somewhere) to re-implement timeslicing in terms of the kernel timeout mechanism.  Then each CPU could have its own timeout expiration and cancel it as needed.  It's just not a big deal, except for this one case with PM interactions.

And I agree, for this release we should just document this.

closed 2021-08-16 by github-actions[bot]