zephyrproject-rtos/zephyr #33491: Fwrite function will cause the program to crash when wrong parameters passed

xiaofst, 2021-03-19:
**Describe the bug**
I am testing the libc module in qemu_x86, It has a function fwrite.
when I passed the wrong  parameters, the program crashed.

**To Reproduce**
Steps to reproduce the behavior:
1. write the code "fwrite("This 3", -1, 1, stdout);"
2. twister -p qemu_x86 -T tests/lib/c_lib/
3. See error

**Expected behavior**
program runs normally

**Impact**
Looking to move forward on working with the fwrite function

**Logs and console output**
^[[0mSeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.5.0-751-gee56f3c081b8  ***
Running test suite test_c_lib
===================================================================
START - test_limits
This 3^@(long_max + long_one == LONG_MIN) is false^@exit failed^@strtok_r error '%s' / '%s'^@(len not equal to tlen)^@(len equal to tlen)^@1,2,3,4,5^@, ^@,, 1 ,2  ,3   4,5  ^@1,2 3,,,4 5  ^@0,1,,,2 3,,,4 5^@A,,,2,3,,,4 5^@1,,,2,3,,,^@| ^@1|2|3,4|5^@test_c_lib^@2^@4^@test_limits^@test_ssize_t^@test_stdbool^@test_stddef^@test_stdint^@test_memcmp^@test_strchr^@test_strcpy^@test_strncpy^@test_memset^@test_strlen^@test_strcmp^@test_strxspn^@test_bsearch^@test_abs^@test_atoi^@test_strncmp^@test_strtoxl^@test_checktype^@test_memstr^@test_time^@test_abort^@test_exit^@test_str_operate^@test_tolower_toupper^@test_strtok_r^@%c: syscall %s failed check: Memory region %p (size %zu) %s access denied
^@WEST_TOPDIR/zephyr/lib/os/heap.c^@chunk_used(h, c)^@ASSERTION FAIL [%s] @ %s:%d
^@      unexpected heap state (double-free?) for memory at %p
^@left_chunk(h, right_chunk(h, c)) == c^@       corrupted heap bounds (buffer overflow?) for memory at %p
^@(align & (align - 1)) == 0^@  align must be a power of 2
^@bytes > heap_footer_bytes(bytes)^@    heap size is too small
^@buf_sz > chunksz(sizeof(struct z_heap))^@chunk0_size + min_chunk_size(h) < buf_sz^@(nil)^@write^@%c: ^@%02x ^@   ^@|^@%c^@UART_0^@WEST_TOPDIR/zephyr/drivers/interrupt_controller/intc_system_apic.c^@irq <= ((24 + 6) - 1)^@ invalid irq line
^@sys_clock^@WEST_TOPDIR/zephyr/include/spinlock.h^@z_spin_lock_valid(l)^@      Recursive spinlock %p
^@z_spin_unlock_valid(l)^@      Not my spinlock %p
^@%c: EAX: 0x%08x, EBX: 0x%08x, ECX: 0x%08x, EDX: 0x%08x
^@%c: ESI: 0x%08x, EDI: 0x%08x, EBP: 0x%08x, ESP: 0x%08x
^@%c: EFLAGS: 0x%08x CS: 0x%04x CR3: 0x%08lx
^@%c: call trace:
^@%c: EIP: 0x%08x
^@%c: NULL base ptr
^@%c: unaligned frame ptr
^@%c:      corrupted? (bp=%p)
^@%c:      0x%08lx (0x%lx)
^@%c: Divide by zero
^@%c: Debug
^@%c: Non-maskable interrupt
^@%c: Breakpoint
^@%c: Overflow
^@%c: Bound range exceeded
^@%c: Invalid opcode
^@%c: Floating point unit device not available
^@%c: Double fault (code 0x%lx)
^@%c: Co-processor segment overrun
^@%c: Invalid TSS (code 0x%lx)
^@%c: Segment not present (code 0x%lx)
^@%c: Stack segment fault
^@%c: General protection fault (code 0x%lx)
^@%c: x87 floating point exception
^@%c: Alignment check (code 0x%lx)
^@%c: Machine check
^@%c: SIMD floating point exception
^@%c: Virtualization exception
^@%c: Security exception
^@execute^@user^@supervisor^@%c: Page fault at address %p (error code 0x%lx)
^@%c: Reserved bits set in page tables
^@%c: Linear address not present in page tables
^@%c: Access violation: %s thread not allowed to %s
^@%c: Protection key disallowed
^@%c: SGX access control violation
^@%s^@WEST_TOPDIR/zephyr/arch/x86/core/x86_mmu.c^@(src[i] & (1ULL << (7))) == 0^@       large page encountered
"~/zephyrproject/zephyr/twister-out/qemu_x86/tests/lib/c_lib/libraries.libc/handler.log" [dos] 139L, 9719C    

**Environment (please complete the following information):**
 - Linux
 - Toolchain Zephyr SDK 0.12.3

carlescufi, 2021-03-23:
@xiaofst are you using the minimal C library or newlib?

enjiamai, 2021-03-23:
@carlescufi got it, will take a look as soon. Look like need to add some checking code for it.

enjiamai, 2021-04-23:
@carlescufi  sorry for reply late, after checking it, I found this is due to:
When we call "fwrite("This 3", -1, 1, stdout);",  the size_t -1 means 4294967295, so actually parameter is:
```
Breakpoint 2, fwrite (ptr=0x1130a1, size=4294967295, nitems=1, stream=0x2) at zephyrproject/zephyr/lib/libc/minimal/source/stdout/stdout_console.c:102
102             return zephyr_fwrite(ptr, size, nitems, stream);
``` 
And this incorrect size cannot be detected by neither userspace z_vrfy_zephyr_fwrite()
```
 Z_OOPS(Z_SYSCALL_MEMORY_ARRAY_READ(ptr, nitems, size));
```
nor z_impl_zephyr_fwrite() itself,  so the fwrite() works, and overflow happens.

I am thinking if we should check "the size * nitems  cannot greater than string length".

enjiamai, 2021-04-23:
> @xiaofst are you using the minimal C library or newlib?

In this case, it is using the minimal C library.

pabigot, 2021-05-06:
See https://github.com/zephyrproject-rtos/zephyr/pull/34531#pullrequestreview-653659547: the C library is doing exactly what it was told to do.

de-nordic, 2021-05-07:
It is not a bug; the `size`, within context of `fwrite` definition,  is of type `size_t`, so the value -1 is casted as expected.
It is quite common trick  to give unsigned type the highest value by assigning it `-1`.

Internally, within the  `fwrite` function expects `size_t` which means that any value that fits within range of the type is correct.
Lets for example assume that  `size_t` is 16 bit, then -1 is equivalent of 65535 or  0xFFFF, and both values seem reasonable: maybe somebody is trying to dump entire 16bit address space from 0 to -1, by one byte, via stdout.

The correctness of `size` and `nitems` should be ensured by caller, it is not possible to predict intentions of developer or meaning of these numbers within the function.

enjiamai, 2021-05-07:
Hi @carlescufi @nashif , the point from @pabigot and @de-nordic is totally making sense, it's not a bug. Can we close this issue? thanks!

closed 2021-05-07 by carlescufi