zephyrproject-rtos/zephyr #33505: WS2812 SPI LED driver with DMA on nrf52 bad SPI data

clementlgl, 2021-03-19:
**Describe the bug**
I'm trying to make the WS2812 LED example work on nrf52dk_nrf52832.

WS2812 LED driver does not work properly on nrf52. When I use the SPI DMA driver (nordic,nrf-spi), a low level is maintained on the line for a few µS in the middle of the frame (see diagram below) and the LEDs do not support it.

[image]

**Expected behavior**
If I use the standard SPI driver (nrf-spi) the bits are less regular but there is no low level maintained several microseconds in the middle of the frame. And it works :tada: 

[image]

I think the problem comes from the SPI driver in DMA mode, have you ever had this problem?

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52dk_nrf52832 -p auto samples/drivers/led_ws2812/
2. west flash

**Impact**
The LEDs don't work and using the spi driver without DMA is not a good solution.

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain zephyr 0.12.3

peterniebert, 2021-03-20:
I would defintely say that this is not caused by a bug of zephyr, rather it is a design issue in nRF SOCs

I have observed the same thing some time ago independently in the context of mBed OS. Some not obvious change in the processor state must cause this pause. Or maybe, it is a priority issue on the bus?
You might try to place the buffer in a predefined memory page that is not used by anything else?

 Maybe this should be discussed with engineers at devzone? 

On the other hand, most WS2812 like LEDs now accept pauses of this dimension so it should not be a problem with modern LEDs. The downside is that reset times are now in the low 100s of us, which may negatively affect the frame rate. So you could still solve the problem by changing to modern LEDs.

peterniebert, 2021-03-20:
Reset times:

Take a look here: https://datasheet.lcsc.com/szlcsc/2006151006_Worldsemi-WS2812B-B_C114586.pdf
"RESET time>280μs, it won't cause wrong reset while interruption, its upports the lower frequency and inexpensive MCU."

clementlgl, 2021-03-22:
Thank you @peterniebert, 

Indeed I tried with LEDs bought more recently and it works well. My other strip is too old ... 

If it's a SOC bug it's a shame but I think we can close this issue. I also use Mbed OS and I will try on it.

Thanks for the link to the  WS2812 PDF, I had read it too quickly but I had not thought of an evolution of the WS2812 and that the LEDs of my old strip were old !

Thanks for your help

carlescufi, 2021-03-23:
Closing this since it doesn't seem to be a Zephyr issue. @anangl please reopen if you think this is an actual bug.

closed 2021-03-23 by carlescufi