zephyrproject-rtos/zephyr #33508: Location/tagging for `example-application` manifest repo issues?

gregshue, 2021-03-19:
The new `example-application` repository represents a new class of repositories (see below). This may affect where and how repository issues are tracked.  Where should the issues be kept and how should the be tagged?

The `zephyr` repository is necessarily **independent of** and **depended upon by** `example-application`.  which means:

1. The `example-application` must never appear appear in the `zephyr` default manifest.
2. The `example-application` repository will have a manifest dependent on `zephyr` default and/or release manifests.
3. The `example-application` repository may have separate branches for code that works with:

    - the head of the `zephyr` repository
    - the latest `zephyr` release
    - the head of each  actively supported `zephyr` LTS branch

4. The release-readiness metrics generated from `zephyr` issues must not be affected by `example-application` issues.
5. The `example-application` is an "external" Zephyr module, which should have its own versioning and release cycle.
6. A `zephyr` bug in the module extension mechanism may reference an `example-application` PR to demonstrate the issue.

Preferences?

gregshue, 2021-07-07:
Subsequent decisions:
- Keep `example-application` issues within the `zephyr` issues list
- `example-application` will be released `zephyr`
- `example-application release tags will match `zephyr`, beginning with `v2.6.0`
- The `example-application` manifest in the `main` branch will import the latest `zephyr` manifest. Releases of `example-application` will have a manifest referencing the release version of `zephyr`.

Pending question(s):
- What label should be applied to an issue to mark it as being relevant to `example-application`?

nashif, 2024-02-08:
I do not think any of that warrant an enhancement issue, if there are bugs or whatever in the example-application, we will deal with and decide to add labels once they accumulate.

closed 2024-02-08 by nashif