zephyrproject-rtos/zephyr #33512: build: build target is always out-of-date

e-rk, 2021-03-19:
**Describe the bug**
The build target is always considered by the build system as out-of-date, which forces the link stage to be executed each time `west build/debug/attach/flash` is invoked. The link stage of some applications may take more than a few seconds and commands that used to execute instantaneously now take much longer.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/basic/blinky`
2. `west build -b nrf52840dk_nrf52840 -p`
3. `west build`
4. Observe that the link stage is repeated the second time `west build` is invoked.

**Expected behavior**
`ninja` should do nothing the second time around.
```
$ west build
ninja: no work to do.
```

**Impact**
annoyance

**Logs and console output**
```
$ west build
[1/1] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       17500 B         1 MB      1.67%
            SRAM:        5536 B       256 KB      2.11%
        IDT_LIST:          0 GB         2 KB      0.00%

$ west build
[1/1] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       17500 B         1 MB      1.67%
            SRAM:        5536 B       256 KB      2.11%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: `arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)`
 - CMake: `3.19.5`
 - ninja: `1.8.2`
 - Commit SHA: 856ff7db3d0bf5b4c3b83bb6a716803346584bbc

**Additional context**
`git bisect` points to c4c79f5f334ef03e373c85231bbffdc82240933a

carlescufi, 2021-03-19:
I've observed this as well, and also dislike it. Thanks for creating the issue.

tejlmand, 2021-03-22:
Thanks for creating this issue.
Seems to happen because a byproduct is being renamed, causing ninja to not find it.

This again leads ninja to relink, in order to create the missing file, which then again is being renamed.
https://github.com/zephyrproject-rtos/zephyr/blob/90d7322d8cc09072b096b146193917fa2c2d5f3f/CMakeLists.txt#L1119

tejlmand, 2021-03-22:
Fixed: #33571

PR #33571 (2021-03-22): cmake: removing map file from BYPRODUCTS as it is being renamed
Fixes: #33512

The ${ZEPHYR_FINAL_EXECUTABLE}.map file is being renamed at a later
stage.

The renamed, and thus missing file, causes ninja to relink the target
in order to re-create the map file, which then again is renamed.

This is fixed by removing the ${ZEPHYR_FINAL_EXECUTABLE}.map as a
byproduct and only has the byproduct on the renamed file to ensure
proper cleanup, see: #23449.
commit 55e3ab79 (2021-03-22): cmake: removing map file from BYPRODUCTS as it is being renamed

Fixes: #33512

The ${ZEPHYR_FINAL_EXECUTABLE}.map file is being renamed at a later
stage.

The renamed, and thus missing file, causes ninja to relink the target
in order to re-create the map file, which then again is renamed.

This is fixed by removing the ${ZEPHYR_FINAL_EXECUTABLE}.map as a
byproduct and only has the byproduct on the renamed file to ensure
proper cleanup, see: #23449.
closed 2021-03-22 by nashif