zephyrproject-rtos/zephyr #33515: arm64/mmu: Are you sure it's OK to use atomic_cas before the MMU is initialized?

luisgerhorst, 2021-03-19:
I've been testing the latest version with my port to the Raspberry Pi 3 Model B+ and had an issue where the MMU would not initialize on the hardware (in qemu it worked find). Using a debugger I discovered that the spinlock in set_mapping is the issue:
https://github.com/zephyrproject-rtos/zephyr/blob/f9a0e20432818308bead889fece60869337a32cb/arch/arm/core/aarch64/mmu/arm_mmu.c#L227
Removing the k_spin_lock/k_spin_unlock it that function fixed it.

Are you sure it's OK to use the compiler's atomics before the MMU is initialized? I just leave this here as a question, it could as well be some misconfiguration in my port. If required I can share the code.

npitre, 2021-03-20:
It most probably isn't.
Will have to think of a clean way to work around that.

BTW, did you consider contributing your RPM board files to Zephyr?

luisgerhorst, 2021-03-20:
OK, thanks for your answer.

I'll have to clean them up but I will definitely think about contributing them. I've pushed them them to https://github.com/luisgerhorst/zephyr/tree/raspi3bp-support

luisgerhorst, 2021-03-21:
Another note: I‘m not sure if this is true only for printk or also LOG_*, but if I remember correctly  these functions may also use a lock in certain configurations. Is that true or am mixing things up here?

npitre, 2021-03-21:
> Another note: I‘m not sure if this is true only for printk or also
> LOG_*, but if I remember correctly these functions may also use a lock
> in certain configurations. Is that true or am mixing things up here?

It is possible, yes. But there is a limit to how much we should
accommodate very early LOG_* before the MMU is on. We try to turn the MMU 
on very early during boot and avoid logging anything by default until then.
If you really need debugging output before that point then you should 
already be familiar with more exotic debugging methods anyway.

carlescufi, 2021-03-22:
> OK, thanks for your answer.
> 
> I'll have to clean them up but I will definitely think about contributing them. I've pushed them to https://github.com/luisgerhorst/zephyr/tree/raspi3bp-support

This contribution would be most welcome @luisgerhorst. There are many Zephyr users out there that do not own AArch64 development boards but that have access to a Raspberry Pi.

npitre, 2021-05-06:
This is solved with commit 28dc807f5057

closed 2021-05-06 by npitre