zephyrproject-rtos/zephyr #33525: ST Nucleo G071RB board support issue

mtudan, 2021-03-20:
**Describe the bug**
Flashing of the ST Nucleo G071RB board cannot be done because you cannot enable the support of the board on pyocd by adding “pack” support - stm32g071rb is probably missing in the new version of the pack "Keil.STM32G0xx_DFP.1.3.0.pack".

**To Reproduce**
Steps to reproduce the behavior:
1. pyocd pack --update
2. pyocd pack --install stm32g071rb

**Expected behavior**
Expect to be able to flash the board.

**Logs and console output**
```
❯ pyocd pack --install stm32g071rb       
0000386:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
```

**Environment (please complete the following information):**
 - OS: Linux
 - master

mtudan, 2021-03-20:
This might be related:
https://github.com/pyocd/pyOCD/issues/638

str4t0m, 2021-03-20:
The issue is related to the download process of the pack, so I suggest you open the issue in the pyocd repo.

Until this is fixed in pyocd you can download it from [keil](https://www.keil.com/dd2/Pack/) and pass the path to your locally downloaded pack either in the board.cmake file or directly in the west flash command. 
`west flash  -- --flash-opt=--pack=/path/to/Keil.STM32G0xx_DFP.1.3.0.pack`

alternatively you can use the stm32cubeprogrammer or jlink runner.
To use the jlink runner you need to upgrade the st-link to an [jlink ob](https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/), or use an external probe.
`west flash -r stm32cubeprogrammer`

erwango, 2021-03-22:
Alternatively, we can also add openocd support, as STM32G0 is now supported by openocd.