zephyrproject-rtos/zephyr #33526: boards:  Optimal way to have customized dts for my project.

Kshitij4kk, 2021-03-20:
I am working on STM32H723GZ MCU with Nucleo-144 board.

My project requires a customized dts .
I found a matching file at zephyr/boards/arm/nucleo_h723gz/nucleo_h723gz.dts.

After going through documentation, I found that I can create an overlay file in my project folder and use it to achieve the customization.
Just wanted to know whether there are other ways of achieving a customized  dts ?

@erwango @KwonTae-young

KwonTae-young, 2021-03-22:
As far as I know, nothing other than `overlay`.  
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/sensor/sht3xd/boards

erwango, 2021-03-22:
> After going through documentation, I found that I can create an overlay file in my project folder and use it to achieve the customization.
Just wanted to know whether there are other ways of achieving a customized dts ?

To answer the question, it would be good to understand the problem you're having with .overlay

Kshitij4kk, 2021-03-23:
@erwango 
No problems observed with overlay method.
I come from Linux background, where I used to write custom dts for the board so just wanted to know whether there exists any method in Zephyr to achieve the same without disturbing the project tree itself.
That being said, I have no issues with using the overlay method and would be happy to proceed with it.

Thanks @KwonTae-young and @erwango for the support.

closed 2021-03-25 by Kshitij4kk