zephyrproject-rtos/zephyr #33529: SSD1306 DTC properties not being generated in devicetree_unfixed.h

jeremyherbert, 2021-03-21:
**Describe the bug**
When building a project with the SSD1306 enabled and set up, I get build errors indicating that devicetree properties are not being generated in the `devicetree_unfixed.h` file, for example:

```
/home/jeremy/nrfconnect/ncs/zephyr/drivers/display/ssd1306.c: In function 'ssd1306_set_timing_setting':
In file included from /home/jeremy/nrfconnect/ncs/zephyr/include/arch/arm/aarch32/arch.h:20,
                 from /home/jeremy/nrfconnect/ncs/zephyr/include/arch/cpu.h:19,
                 from /home/jeremy/nrfconnect/ncs/zephyr/include/kernel_includes.h:33,
                 from /home/jeremy/nrfconnect/ncs/zephyr/include/kernel.h:17,
                 from /home/jeremy/nrfconnect/ncs/zephyr/include/init.h:11,
                 from /home/jeremy/nrfconnect/ncs/zephyr/include/device.h:22,
                 from /home/jeremy/nrfconnect/ncs/zephyr/drivers/display/ssd1306.c:13:
/home/jeremy/nrfconnect/ncs/zephyr/drivers/display/ssd1306.c: In function 'ssd1306_set_timing_setting':
zephyr/include/generated/devicetree_unfixed.h:4093:39: error: 'DT_N_S_soc_S_i2c_40003000_S_ssd1306_3c_P_prechargep' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_i2c_40003000_S_ssd1306_3c_P_reg'?
#define DT_N_INST_0_solomon_ssd1306fb DT_N_S_soc_S_i2c_40003000_S_ssd1306_3c
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:2176:24: note: in definition of macro 'DT_CAT'
 #define DT_CAT(a1, a2) a1 ## a2
                        ^~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1699:34: note: in expansion of macro 'DT_PROP'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                  ^~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/sys/util_internal.h:106:36: note: in expansion of macro 'DT_N_INST_0_solomon_ssd1306fb'
 #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
                                    ^
/home/jeremy/nrfconnect/ncs/zephyr/include/sys/util_internal.h:105:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1677:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1699:42: note: in expansion of macro 'DT_DRV_INST'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                          ^~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/drivers/display/ssd1306.c:120:3: note: in expansion of macro 'DT_INST_PROP'
   DT_INST_PROP(0, prechargep),
   ^~~~~~~~~~~~
zephyr/include/generated/devicetree_unfixed.h:4093:39: note: each undeclared identifier is reported only once for each function it appears in
 #define DT_N_INST_0_solomon_ssd1306fb DT_N_S_soc_S_i2c_40003000_S_ssd1306_3c
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:2176:24: note: in definition of macro 'DT_CAT'
 #define DT_CAT(a1, a2) a1 ## a2
                        ^~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1699:34: note: in expansion of macro 'DT_PROP'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                  ^~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/sys/util_internal.h:106:36: note: in expansion of macro 'DT_N_INST_0_solomon_ssd1306fb'
 #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
                                    ^
/home/jeremy/nrfconnect/ncs/zephyr/include/sys/util_internal.h:105:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1677:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/include/devicetree.h:1699:42: note: in expansion of macro 'DT_DRV_INST'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                          ^~~~~~~~~~~
/home/jeremy/nrfconnect/ncs/zephyr/drivers/display/ssd1306.c:120:3: note: in expansion of macro 'DT_INST_PROP'
   DT_INST_PROP(0, prechargep),
   ^~~~~~~~~~~~

```

**To Reproduce**
Add the following file as `display.overlay` to a blank project in the project root directory:

```
&i2c0 {
    status = "okay";
    clock-frequency = < 400000 >;

    ssd1306@3c {
        compatible = "solomon,ssd1306fb";
        reg = <0x3c>;
        label = "SSD1306";
        width = <128>;
        height = <64>;
        segment-offset = <0>;
        page-offset = <0>;
        display-offset = <0>;
        multiplex-ratio = <63>;
        segment-remap;
        com-invdir;
        prechargep = <0x22>;
    };
};
```

edit prj.conf to contain:
```
CONFIG_GPIO=y

CONFIG_ASSERT=y

CONFIG_I2C=y

CONFIG_DISPLAY=y
CONFIG_DISPLAY_LOG_LEVEL_ERR=y

CONFIG_SSD1306=y
CONFIG_SSD1306_SH1106_COMPATIBLE=y

CONFIG_LOG=y

CONFIG_LVGL=y
CONFIG_LVGL_USE_THEME_MONO=y

CONFIG_LVGL_USE_LABEL=y
CONFIG_LVGL_USE_BAR=y
CONFIG_LVGL_USE_LINEMETER=y
CONFIG_LVGL_USE_GAUGE=y

CONFIG_LVGL_COLOR_DEPTH_1=y
CONFIG_LVGL_BITS_PER_PIXEL=1
CONFIG_LVGL_HOR_RES_MAX=128
CONFIG_LVGL_VER_RES_MAX=64
CONFIG_LVGL_DPI=130
CONFIG_LVGL_VDB_SIZE=16
```

Steps to reproduce the behavior:
1. Build using `DTC_OVERLAY_FILE=display.overlay BOARD=adafruit_feather_nrf52840 west build` in the project folder

**Expected behavior**
The devicetree properties should be generated in the header file.

**Impact**
Showstopper, code doesn't build.

**Environment (please complete the following information):**
 - OS: Linux ubuntu 5.4.0-66-generic
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 4ce908a2342494c19df09f347ce79ab81b415185

jfischer-no, 2021-03-22:
i2c node of adafruit_feather_nrf52840 board is missing compatible property, "nordic,nrf-twim" or "nordic,nrf-twi"

```diff c
diff --git a/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840.dts b/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840.dts
index bb142eb9b6..4ca4c78b15 100644
--- a/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840.dts
+++ b/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840.dts
@@ -76,6 +76,7 @@
 };
 
 &i2c0 {
+       compatible = "nordic,nrf-twim";
        status = "okay";
        sda-gpios = <&gpio0 12 0>;
        scl-gpios = <&gpio0 11 0>;
```

jeremyherbert, 2021-03-22:
Thank you! I don't think I would have been able to find that one myself.

galak, 2021-03-22:
Closing thee issue as it seems @jfischer-no comment resolves this.

jeremyherbert, 2021-03-22:
@galak I don’t think it should be closed, this should really be fixed in the board definition file for the adafruit board.

jeremyherbert, 2021-03-22:
Also I’m not sure that the SSD1306 driver should necessarily depend on compatibility with a Nordic implementation, seeing as they are not related.

galak, 2021-03-22:
@jeremyherbert I dont think the SSD1306 driver depends on compatibility with a nordic implementation.  I think the issue is that there i2c controller devicetree isn't configured properly and thus the SSD1306 driver can't build as it can't find a proper i2c controller.

jeremyherbert, 2021-03-22:
Ah, that makes sense. 

I will create a pull request with this updated line in the next 24h.

closed 2021-03-25 by carlescufi