zephyrproject-rtos/zephyr #33539: up_squared: tests/kernel/mem_heap/mheap_api_concept failed.

chen-png, 2021-03-22:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/mem_heap/mheap_api_concept
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.5.0-1430-g911ba334a06a 
Running test suite mheap_api

START - test_mheap_malloc_free
PASS - test_mheap_malloc_free in 0.1 seconds

START - test_mheap_calloc
PASS - test_mheap_calloc in 0.1 seconds

START - test_mheap_malloc_align4
PASS - test_mheap_malloc_align4 in 0.1 seconds

**START - test_mheap_threadsafe
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_concept.c:61: tmheap_handler: (block[thread_id] is NULL)
memory is not allocated
FAIL - test_mheap_threadsafe in 0.18 seconds**

START - test_k_aligned_alloc
PASS - test_k_aligned_alloc in 0.1 seconds

START - test_sys_heap_mem_pool_assign
PASS - test_sys_heap_mem_pool_assign in 0.1 seconds

START - test_malloc_in_isr
PASS - test_malloc_in_isr in 0.1 seconds

START - test_malloc_in_thread
PASS - test_malloc_in_thread in 0.1 seconds

Test suite mheap_api failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora33
 - Toolchain: zephyr-sdk-0.12.3
 - Commit ID: 911ba334a06a

chen-png, 2021-03-22:
also failed on ehl_crb board.

yerabolu, 2021-05-12:
Update: Test is not failing on UP2 but it is still failing on EHL. So changed the title accordingly.

andyross, 2021-05-12:
up_squared switched its timer driver.  In theory that shouldn't actually change the timing of OS tick delivery (10khz is much slower than either 19.2 MHz or 1.5 GHz) but in practice very small aliasing can change behavior of what is presumptively a race condition in the test.

andyross, 2021-05-14:
This doesn't reproduce anymore.  Looks like it was fixed by https://github.com/zephyrproject-rtos/zephyr/commit/f100566f7a3397bb0e2f168853dceeb4f24ac188 by @IRISZZW which merged on Wednesday.

Indeed, that bug was holding a thread pointer in static storage shared between two threads racing against each other.  I'm shocked it worked on any SMP system.  It certainly was going to fail on x86 where the memory model prevents the hardware from hiding races behind the memory ordering.

closed 2021-05-14 by andyross