zephyrproject-rtos/zephyr #33543: ehl_crb: tests/subsys/edac/ibecc failed.

chen-png, 2021-03-22:
**Description**
the three testcases "edac.ibecc.production", "edac.ibecc.injection" and " edac.ibecc.injection.user" all failed on ehl_crb board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b ehl_crb -p auto tests/subsys/edac/ibecc 
    -DCONFIG_EDAC_ERROR_INJECT=n  -DCONFIG_EDAC_LOG_LEVEL_DBG=y
2. copy binary to USB/PXE boot media
3. boot
4. See error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.5.0-1430-g911ba334a06a
Running test suite ibecc

START - test_ibecc_initialized
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/edac/ibecc/src/ibecc.c:30: test_ibecc_initialized: (dev is NULL)
Device not found
FAIL - test_ibecc_initialized in 0.14 seconds

START - test_ibecc_api
E: Page fault at address 0x10 (error code 0x0)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to read
E: PTE: not present
E: RAX: 0x0000000000000000 RBX: 0x00000000005a0028 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x000000000058e3c0 RSP: 0x000000000058e3c0
E:  R8: 0x0000000000103645  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x000000000010b900 R13: 0x0000000000000247 R14: 0x000000000010b900 R15: 0x0000000000000000
E: RSP: 0x000000000058e3c0 RFLAGS: 0x0000000000010202 CS: 0x0018 CR3: 0x0000000000ba5000
E: RIP: 0x000000000010010b
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10aee0 (ztest_thread)
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora33
 - Toolchain: zephyr-sdk-0.12.3
 - Commit ID: 911ba334a06a

finikorg, 2021-03-22:
I have tested today upstream Zephyr on our EHL, they works fine. What is your SKU?

finikorg, 2021-03-22:
I think you have SKU11, this should fix the issue.

chen-png, 2021-03-23:
> What is your SKU?

I tested on SKU11.

chen-png, 2021-03-23:
this patch #33564 only fixed the testcase "edac.ibecc.production", for "edac.ibecc.injection" and " edac.ibecc.injection.user", they were still failed.

result of "edac.ibecc.injection":
`Booting Zephyr OS build zephyr-v2.5.0-1469-g76f35f29f01e 
Running test suite ibecc

START - test_ibecc_initialized
Test ibecc driver is initialized
PASS - test_ibecc_initialized in 0.4 seconds

START - test_ibecc_api
PASS - test_ibecc_api in 0.1 seconds

**START - test_ibecc_error_inject_api
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/edac/ibecc/src/ibecc.c:98: test_ibecc_error_inject_api: (val not equal to TEST_ADDRESS1)
Read back value differs
FAIL - test_ibecc_error_inject_api in 0.17 seconds**

START - test_ibecc_error_inject_test_cor
Mapped 0x1000 to 0xbffe5000
Read value 0xbffe5000: 0x5016010f
Wrote value 0xface at 0xbffe5000
Read value 0xbffe5000: 0xface
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/edac/ibecc/src/ibecc.c:179: test_inject: (num_int equal to 0)
Interrupt handler did not execute
FAIL - test_ibecc_error_inject_test_cor in 0.128 seconds

START - test_ibecc_error_inject_test_uc
Mapped 0x2000 to 0xbffe4000
Read value 0xbffe4000: 0x1812ad
Wrote value 0xface at 0xbffe4000
Read value 0xbffe4000: 0xface
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/edac/ibecc/src/ibecc.c:179: test_inject: (num_int equal to 0)
Interrupt handler did not execute
FAIL - test_ibecc_error_inject_test_uc in 0.128 seconds

Test suite ibecc failed.

PROJECT EXECUTION FAILED`

finikorg, 2021-03-23:
@chen-png Have you enabled error injection in BIOS needed for injecting errors for this test case?

finikorg, 2021-03-23:
@nashif @jhedberg I can have a check verifying that after write to error injection register I can read back. So I can detect that error injection is not supported and can skip those remaining tests, so we would not get failed cases.

finikorg, 2021-03-23:
I probably need to add some information to README

chen-png, 2021-03-23:
> Have you enabled error injection in BIOS needed for injecting errors for this test case?

No, I don't know about this, maybe a README is good.

finikorg, 2021-03-23:
@chen-png  https://github.com/zephyrproject-rtos/zephyr/pull/33643

chen-png, 2021-03-25:
closed this, it has been fixed after bios configuration in #33643

PR #33564 (2021-03-22): edac: ibecc: Add supported SKU device IDs
List all supported device IDs found in External Design Specification
Volume 1 which have IBECC supported.

Fixes #33543
commit 365be454 (2021-03-22): edac: ibecc: Add supported SKU device IDs

List all supported device IDs found in External Design Specification
Volume 1 which have IBECC supported.

Fixes #33543
closed 2021-03-25 by chen-png