zephyrproject-rtos/zephyr #33545: ehl_crb: tests/arch/x86/info failed.

chen-png, 2021-03-22:
**Description**
nothing output.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b ehl_crb -p auto tests/arch/x86/info
2. copy binary to USB/PXE boot media
3. boot
4. See error

**Logs and console output**
nothing from console output

**Expected**
should see testcases results from console.

**Environment (please complete the following information):**
 - OS: Fedora33
 - Toolchain: zephyr-sdk-0.12.3
 - Commit ID: 911ba334a06a

jenmwms, 2021-03-25:
I cannot reproduce this issue. I tested at the reported commit revision 911ba33 and again at more recent revision a9b4cae0 in case it has been resolved. Both configurations Pass for me using twister.

@chen-png @jhedberg Are you able to reproduce this reported issue?

jenmwms, 2021-03-25:
Oh interesting, I do not see it with SDK 12.1 but I do see it with SDK 12.3. I'll keep testing and see what we find.

jhedberg, 2021-03-25:
@jenmwms I haven't tried with twister, but manually building the info sample and booting the zephyr.efi through PXE (on `ehl06` in our lab, FWIW) gets stuck at this point:
```
*** Zephyr EFI Loader ***
Zeroing 4774384 bytes of memory at 0x106000
Copying 32768 data bytes to 0x1000 from image offset
Copying 24576 data bytes to 0x100000 from image offset 32768
Copying 6317568 data bytes to 0x593a00 from image offset 57344
Jumping to Entry Point: 0x1123 (48 31 c0 48 31 d2 48)
```

E.g. hello_world using the same build process and machine works fine:
```
Jumping to Entry Point: 0x10f7 (48 31 c0 48 31 d2 48)
*** Booting Zephyr OS build zephyr-v2.5.0-1589-g8965df96222e  ***
Hello World! ehl_crb
```

jenmwms, 2021-03-25:
@jhedberg Thanks for the feedback! I will sort out my env issues asap.

jenmwms, 2021-03-26:
I can reproduce the issue now, I'll work on bisect and debug to update here soon.

jenmwms, 2021-04-14:
Noting that this is still an issue. This one is not simple to bisect based on last known working point in history, unknown when that was so moving forward with debug otherwise.

jenmwms, 2021-05-14:
@chen-png Do you know if you this works on other 64-bit (up_squared)? 
FWIW - Was hoping to test against an emulator if possible, but this test allows qemu_x86 and not qemu_x86_64 so it fails. Shouldn't it work for qemu_x86_64 if it works for 64-bit boards like ehl_crb and up_squared?

andyross, 2021-05-14:
Indeed, this hangs immediately after the boot banner (and before entering main()) on qemu_x86_64.  My guess is this is just bitrot with 64 bit support for multiboot or something.

jenmwms, 2021-05-15:
Re-tested with that PR above - it resolves the issue.

chen-png, 2021-05-17:
>Do you know if you this works on other 64-bit (up_squared)?

it works on up_squared.

PR #35314 (2021-05-15): arch/x86: Correct multiboot interpretation when building for EFI
When loaded via EFI, we obviously don't have a multiboot info pointer
available (we might have an EFI system table, but zefi doesn't pass
that through yet).  Don't try to parse the "whatever garbage was in
%rbp" as a multiboot table.

The configuration is a little clumsy, as strictly our EFI kconfig just
says we're "building for" EFI but not that we'll boot that way.  And
tests like arch/x86/info are trying to set CONFIG_MULTIBOOT=n
unconditionally, when it really should be something they detect from
devicetree or wherever.

Fixe [... cut]