zephyrproject-rtos/zephyr #33548: xt-xcc does not support deprecated attribute

hongshui3000, 2021-03-22:
**Describe the bug**
file:gcc.h
```
#ifndef __deprecated
#define __deprecated	__attribute__((deprecated))
#endif
```
file:bluetooth.h
```
	BT_LE_SCAN_FILTER_DUPLICATE __deprecated =
		BT_LE_SCAN_OPT_FILTER_DUPLICATE,
```
xt-xcc does not support deprecated attribute.

**Logs and console output**
```
[17/152] Building C object zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/conn.c.obj
FAILED: zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/conn.c.obj
C:\usr\xtensa\XtDevTools\install\tools\RI-2020.5-win32\XtensaTools\bin\xt-xcc.exe -DBUILD_VERSION=zephyr-v2.4.0-1113-g0bcd66ac36d8 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/xtensa/ZC3827 -I../soc/xtensa/ZC3827/include -I../subsys/settings/include -I../subsys/bluetooth -IE:/work/rtos_work/zephyrproject/modules/crypto/tinycrypt/lib/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/zephyr/soc/ZC3827 -isystem ../lib/libc/minimal/include -Os -imacros E:/work/rtos_work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacrosE:/work/rtos_work/zephyrproject/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros E:/work/rtos_work/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/conn.c.obj -MF zephyr\subsys\bluetooth\host\CMakeFiles\subsys__bluetooth__host.dir\conn.c.obj.d -o zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/conn.c.obj -c E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/conn.c
In file included from ../include/bluetooth/conn.h:23,
                 from ../include/bluetooth/hci.h:20,
                 from E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/conn.c:21:
../include/bluetooth/bluetooth.h:1240: error: expected ',' or '}' before '__attribute__'
In file included from ../include/bluetooth/hci.h:20,
                 from E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/conn.c:21:
../include/bluetooth/conn.h:685: error: expected ',' or '}' before '__attribute__'
```
**Environment (please complete the following information):**
 - OS: ( Windows)
 - Toolchain (xt-cc.)

dcpleung, 2021-03-24:
XCC supported deprecated attributes when I changed `hello_world` to following:

```
#include <zephyr.h>
#include <sys/printk.h>

#include <toolchain.h>

__deprecated char *t = "testing\n";

__deprecated void testing(void)
{
        printk(t);
}

void main(void)
{
        printk("Hello World! %s\n", CONFIG_BOARD);

        testing();
}
```

and it compiled without any errors.

hongshui3000, 2021-03-25:
@dcpleung 
Should not support "__attribute__((deprecated))"？

hongshui3000, 2021-03-25:
@dcpleung 
I found out that this attribute does not support adding this attribute
Have you tried adding these in the enum？
```
enum {
	/** Convenience value when no options are specified. */
	BT_LE_SCAN_OPT_NONE = 0,

	/** Filter duplicates. */
	BT_LE_SCAN_OPT_FILTER_DUPLICATE = BIT(0),

	/** Filter using whitelist. */
	BT_LE_SCAN_OPT_FILTER_WHITELIST = BIT(1),

	/** Enable scan on coded PHY (Long Range).*/
	BT_LE_SCAN_OPT_CODED = BIT(2),

	/**
	 * @brief Disable scan on 1M phy.
	 *
	 * @note Requires @ref BT_LE_SCAN_OPT_CODED.
	 */
	BT_LE_SCAN_OPT_NO_1M = BIT(3),

	BT_LE_SCAN_FILTER_DUPLICATE __deprecated =
		BT_LE_SCAN_OPT_FILTER_DUPLICATE,
	BT_LE_SCAN_FILTER_WHITELIST __deprecated =
		BT_LE_SCAN_OPT_FILTER_WHITELIST,
};
```

PR #33743 (2021-03-29): toolchain: xcc: nullify __deprecated macro
XCC does not support using the deprecated attribute for enum
as it is based on old GCC 4.2. So nullify the macro to avoid
compilation errors.

Fixes #33548
commit 21f62173 (2021-03-29): toolchain: xcc: nullify __deprecated macro

XCC does not support using the deprecated attribute for enum
as it is based on old GCC 4.2. So nullify the macro to avoid
compilation errors.

Fixes #33548
closed 2021-03-29 by nashif