zephyrproject-rtos/zephyr #33549: XT-XCC unknown field 'obj' specified in initializer

hongshui3000, 2021-03-22:
**Describe the bug**
file:kernel.h
```
struct k_poll_event {
	/** PRIVATE - DO NOT TOUCH */
	sys_dnode_t _node;

	/** PRIVATE - DO NOT TOUCH */
	struct _poller *poller;

	/** optional user-specified tag, opaque, untouched by the API */
	uint32_t tag:8;

	/** bitfield of event types (bitwise-ORed K_POLL_TYPE_xxx values) */
	uint32_t type:_POLL_NUM_TYPES;

	/** bitfield of event states (bitwise-ORed K_POLL_STATE_xxx values) */
	uint32_t state:_POLL_NUM_STATES;

	/** mode of operation, from enum k_poll_modes */
	uint32_t mode:1;

	/** unused bits in 32-bit word */
	uint32_t unused:_POLL_EVENT_NUM_UNUSED_BITS;

	/** per-type data */
	union {                            -------------------------------------------------
		void *obj;                                                                                        |
		struct k_poll_signal *signal;                                                            |
		struct k_sem *sem;                                                                         |                               
		struct k_fifo *fifo;                                                                            |
		struct k_queue *queue;                                                                  |
	};   --------------------------------------------------------------------------
};

#define K_POLL_EVENT_INITIALIZER(_event_type, _event_mode, _event_obj) \
	{ \
	.poller = NULL, \
	.type = _event_type, \
	.state = K_POLL_STATE_NOT_READY, \
	.mode = _event_mode, \
	.unused = 0, \
	.obj = _event_obj, \--------------------------------------------------------->this
	}

#define K_POLL_EVENT_STATIC_INITIALIZER(_event_type, _event_mode, _event_obj, \
					event_tag) \
	{ \
	.tag = event_tag, \
	.type = _event_type, \
	.state = K_POLL_STATE_NOT_READY, \
	.mode = _event_mode, \
	.unused = 0, \
	.obj = _event_obj, \--------------------------------------------------------->this
	}

```

xt-xcc cannot recognize the obj object 

Just change it to the following 
```
#define K_POLL_EVENT_STATIC_INITIALIZER(_event_type, _event_mode, _event_obj, \
					event_tag) \
	{ \
	.tag = event_tag, \
	.type = _event_type, \
	.state = K_POLL_STATE_NOT_READY, \
	.mode = _event_mode, \
	.unused = 0, \
       { \--------------------------------------------------------->this
	.obj = _event_obj, \--------------------------------------------------------->this
        } \--------------------------------------------------------->this
	}
```

**Logs and console output**
```
[2/120] Building C object zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj
FAILED: zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj 
C:\usr\xtensa\XtDevTools\install\tools\RI-2020.5-win32\XtensaTools\bin\xt-xcc.exe -DBUILD_VERSION=zephyr-v2.4.0-1113-g0bcd66ac36d8 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/xtensa/ZC3827 -I../soc/xtensa/ZC3827/include -I../subsys/settings/include -I../subsys/bluetooth -IE:/work/rtos_work/zephyrproject/modules/crypto/tinycrypt/lib/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/zephyr/soc/ZC3827 -isystem ../lib/libc/minimal/include -Os -imacros E:/work/rtos_work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacrosE:/work/rtos_work/zephyrproject/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros E:/work/rtos_work/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj -MF zephyr\subsys\bluetooth\host\CMakeFiles\subsys__bluetooth__host.dir\hci_core.c.obj.d -o zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj -c E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c: In function 'hci_tx_thread':
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:5217: error: unknown field 'obj' specified in initializer
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:5217: warning: missing braces around initializer
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:5217: warning: (near initialization for 'events[0].<anonymous>')
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c: In function 'bt_le_per_adv_set_param':
E:/work/rtos_work/zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:7164: warning: comparison is always false due to limited range of data type      
[11/120] Building C object zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/gatt.c.obj
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: (Windows)
 - Toolchain (xt-xcc)

closed 2021-03-26 by galak

aisri, 2023-11-01:
sorry to resurface the discussion,

This fix triggers `-Wc99-designator error` with -Wall -Werror on `xt-clang (clang-10)`

OS: Fedora
Toolchain: xt-xcc

Can you suggest alternatives?

dcpleung, 2023-11-01:
Maybe if you give the anonymous struct a name?

aisri, 2023-11-02:
@dcpleung Thank you for coming back.

ya! I did think about that. It will trigger a cascade of changes across the code base. Will that be a good thing do? please advise!

dcpleung, 2023-11-02:
That will need an RFC for this kind of changes to the tree. You can go on Discord to initial the conversation.

aisri, 2023-11-02:
thanks for the tip. let me bring this conversation.

commit 087fb943 (2021-03-26): kernel: fix XCC compilation with k_poll event initializer

XCC (which is based on GCC 4.2) needs the initializer of
one of the union elements to be enclosed in brackets.
So add them.

Fixes #33549