zephyrproject-rtos/zephyr #33557: there is no network interface to work with for  wifi sample for esp32 board

houssine97, 2021-03-22:
Describe the bug
When running the wifi sample on ESP32 , it reports that there is no network interface to work with:

*** Booting Zephyr OS version 2.5.99  ***ource...

[00:00:00.515,000] <err> net_if: There is no network interface to work with!

To Reproduce
Steps to reproduce the behavior:

cd zephyr/samples/net/wifi
west build -b esp32
west flash
See error
Expected behavior
The sample should get to displaying the wifi shell's prompt without error.

sylvioalves, 2021-03-22:
@houssine97 ESP32 has yet no support regarding zephyr's wifi subsystem due to it's non-offloaded implementation. For now, to get ESP32 wifi running you have the option to use the board specific sample as in `samples/boards/esp32/wifi_station`.

houssine97, 2021-03-22:
@sylvioalves i used the board specific samples wifi_station and i get this error :
<wrn> net_if: iface 0x3ffb00e8 is down
[00:00:02.933,000] <err> esp_event: SYSTEM_EVENT_STA_DISCONNECTED
[00:00:04.981,000] <err> esp_event: SYSTEM_EVENT_STA_DISCONNECTED

nashif, 2021-03-22:
IMO if that is expected behavior, we should close this as a non-bug :)

closed 2021-03-22 by nashif