zephyrproject-rtos/zephyr #33559: [bug] pin setting error on frdm_kl25z boards

hakehuang, 2021-03-22:
**Describe the bug**
uart port has no output, bisect found the issue is related to commit #942916e691dff51b7fc099ddf452ef44bd78679c

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_kl25z ..
3. make flash
4. uart port has no output

**Expected behavior**
output
hello world

**Impact**
frdm_kl25z

**Logs and console output**

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 911ba334a06ac55e28856e49a694083974196f8c

hakehuang, 2021-03-22:
@MaureenHelm  @galak seems the need add more dts fixes

```
DEVICE_DT_GET(DT_NODELABEL(porta));
__ASSERT_NO_MSG(device_is_ready(porta));
```

MaureenHelm, 2021-03-22:
@dleach02 or @galak can you take a look? I don't have this board at the moment

galak, 2021-03-23:
> @dleach02 or @galak can you take a look? I don't have this board at the moment

I looked at the code and nothing obvious.  @dleach02 do you have this board?

dleach02, 2021-03-23:
I have this board

dleach02, 2021-03-23:
bisect information was pointing to the wrong sha. I've updated the sha in the description above.

The problem appears to be a different value is ending up in the clkdiv1. Previously this was hard coded to 0x10010000 and now with the integration of device tree stuff it is set to 0x90000000

dleach02, 2021-03-23:
I have a fix and will get the PR in soon

PR #33648 (2021-03-24): dts: arm: nxp: Fix CORE and Flash clock divider values
In the change to DTS driven clock divider values wrong values
were used in the dts file.

Fixes: #33559
commit cf67a6ee (2021-03-24): dts: arm: nxp: Fix CORE and Flash clock divider values

In the change to DTS driven clock divider values wrong values
were used in the dts file.

Fixes: #33559
closed 2021-03-24 by hakehuang