zephyrproject-rtos/zephyr #33567: sof: framework is redefnining MAX, MIN to version with limited capabilities

nordic-krch, 2021-03-22:
**Describe the bug**
SOF is redefining `MAX`, `MIN` macros to macros with single evaluation of argument. However, such macros cannot be used in certain scenarios, e.g. in `BUILD_ASSERT`. Include with redefinition is used by the wrapper `modules/audio/sof/zephyr/wrapper.c` which is using zephyr code base. If anywhere in this file `MAX` is used in a way not supported by redefined macro compilation fails.

**To Reproduce**
1. Compile `samples/subsys/audio/sof` (e.g. `intel_adsp_cavs25` board)
2. Add `BUILD_ASSERT(MAX(1, 0), "Test build assert");` somewhere in `modules/audio/sof/zephyr/wrapper.c`
3. Compile again and observe that compilation fails.
4. Add following code above and observe that compilation is successful again
```
#undef MAX
#define MAX(a, b) (((a) > (b)) ? (a) : (b))
```

**Expected behavior**
Code should compile, any macro redefinition within zephyr code space should provide equivalent functionality.

**Impact**
Sample fail to compile.

**Logs and console output**
```
/home/krch/work/modules/audio/sof/zephyr/../src/include/sof/math/numbers.h:31:19: error: braced-group within expression allowed only inside a function
   31 | #define MAX(a, b) ({  \
      |                   ^
/home/krch/work/modules/audio/sof/zephyr/wrapper.c:34:14: note: in expansion of macro 'MAX'
   34 | BUILD_ASSERT(MAX(1, 0), "Test build assert");
      |              ^~~
/home/krch/work/modules/audio/sof/zephyr/../src/include/sof/math/numbers.h:31:19: error: expression in static assertion is not an integer
   31 | #define MAX(a, b) ({  \
      |                   ^
/home/krch/work/modules/audio/sof/zephyr/wrapper.c:34:14: note: in expansion of macro 'MAX'
   34 | BUILD_ASSERT(MAX(1, 0), "Test build assert");
```

**Environment (please complete the following information):**
 - Commit SHA 911ba334a0

**Additional context**
It is a blocker for #31535 since CI fails on that sample.

nashif, 2021-03-22:
@lgirdwood  FYI

lgirdwood, 2021-03-22:
@marc-hb can you take a look at this, thanks.

nordic-krch, 2021-04-02:
Any news there? I currently used following workaround: https://github.com/nordic-krch/sof/commit/2f8ef204454d278f0c189e8c7cc70e84f745561a

lgirdwood, 2021-04-02:
@marc-hb any update ? Fwiw @lyakh and @andyross spotted a few others and we need to align SOF to use the Zephyr native version of all common and all RTOS macros. We can have a temporary ```sof-rtos.h``` header that can home the SOF versions of these macros until xtos is removed (this new header should also fail to build is ZEPHYR is defined).

marc-hb, 2021-04-07:
Join the party in https://github.com/thesofproject/sof/pull/4002 if you think you can help.

marc-hb, 2021-04-30:
> Join the party in thesofproject/sof#4002 if you think you can help.

That Z_MAX attempt did not fly.

Sorry for the delay, there was some debate on the SOF changes required and then I got caught into something else.

I see you pointed zephyr/west.yml at your workaround in the mean time: https://github.com/nordic-krch/sof/commit/2f8ef204454d278f0c18

The final SOF decision is to drop the fancy, run-time gcc statement-expressions and align SOF with the basic, compile-time, unsafe and usual MIN and MAX macros used by Zephyr, NOT with the fancier Z_MAX macros. I just submitted that at https://github.com/thesofproject/sof/pull/4119

closed 2021-05-03 by lgirdwood