zephyrproject-rtos/zephyr #33572: <err> esp_event: SYSTEM_EVENT_STA_DISCONNECTED for wifi sample for esp32 board

houssine97, 2021-03-22:
**Describe the bug**
When running the wifi sample on ESP32 , it reports that : 
iface 0x3ffb0140 is down
[00:06:29.873,000] <err> esp_event: SYSTEM_EVENT_STA_DISCONNECTED

**To Reproduce**

Steps to reproduce the behavior:
cd zephyr/samples/boards/esp32/wifi_station
west build -b esp32
west flash
See error
Expected behavior
get ESP32 wifi running

sylvioalves, 2021-03-23:
@houssine97, let me ask you a few things:
1) did you set proper wifi SSID and PASS on the `prj.con` or in the application code?
2) Was it connected before and it suddenly disconnected/iface down?

houssine97, 2021-03-23:
thank you i did not set wifi ssid and pass when i did that it worked .

sylvioalves, 2021-03-23:
Great. let me know if you have any issues. Closing this.

closed 2021-03-23 by sylvioalves