zephyrproject-rtos/zephyr #33573: JSON_OBJ_DESCR_ARRAY_ARRAY is dangerously broken

M1cha, 2021-03-22:
**Describe the bug**
when using JSON_OBJ_DESCR_ARRAY_ARRAY, the subarray is has the token type `JSON_TOK_LIST_START`, but then assigns to the `object` part of the union. `arr_parse` then takes the offset of the array-object (which has nothing todo with the list) treats it as relative to the parent object, and stores the length of the subarray in there.

For the unittest that means that the subarray length(=1) is stored where the name-pointer of the first element would be.
And in fact, the unittest only verifies the names of the second and third elements and just ignores the first one(which would fail the unit tests.

carlescufi, 2021-03-23:
@M1cha thanks for the report. The JSON subsystem is currently orphaned, so is there a chance you could send a PR fixing this?

M1cha, 2021-03-23:
that's sad because it's the best C JSON library I've ever seen. it's very efficient because it doesn't need any malloc or token storage and very convenient because it parses data from/to structs.
But I guess that most people don't use JSON on MCUs due to it's complex an inefficient nature.

I currently don't need array-arrays - I just noticed the bug while trying to understand the code but if I ever feel like fixing it I'll let you know.

martinjaeger, 2021-09-16:
Looks like this bug has been fixed in https://github.com/zephyrproject-rtos/zephyr/pull/36340

@M1cha please link PRs to issues next time, e.g. by adding `Fixes #issue-id` to the commit message. Or close the issue manually after the PR was merged.
I went through the list of issues to support with bug-fixing for next release and it took me a while to realize that this one was already fixed.

closed 2021-09-16 by martinjaeger