zephyrproject-rtos/zephyr #33589: SSD1306 driver no longer works for I2C displays

jeremyherbert, 2021-03-23:
**Describe the bug**
I currently am unable to communicate with an SSD1306 device that works with an older version of Zephyr (approximately mid-2020). Unfortunately I no longer have the exact zephyr version for this.

I have examined the behaviour of the code running on an old device and the current zephyr version with a logic analyser using the exact same SSD1306 display (I moved it between devices). On the old device it works fine, but with the latest zephyr version it no longer works. I also tried several different SSD1306 displays and they all had the same issue.

To communicate with the SSD1306 displays over I2C, the datasheet says that one must send a control byte, and then a data byte repeatedly. See page 20 here: https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf

In this commit: fd2087e080fb3a90132dca3c5c2bb228629ce170 the sending of control bytes appears to be deliberately removed except for the very first one, presumably to make the code work for SPI displays. However, this has broken i2c support in my case.

Also, the use of the `i2c_burst_write` function appears to insert a repeated start condition. This might still work but it is not mentioned in the datasheet as supported; it expects start and stop only.

If I revert all of the changes in that commit above, the display will then turn on, but all I see is noise.

I also need to adjust the code to replace the line here with a non-burst write: https://github.com/zephyrproject-rtos/zephyr/blob/e18fcbba5a4e63b54116f40ae5f0323d5237d329/drivers/display/ssd1306.c#L241 as the burst write causes a repeated start to occur, and this confuses the controller. Perhaps in the past this was not the case?

Once I make those changes, the display works correctly as expected.

**To Reproduce**
`prj.conf`:
```
CONFIG_GPIO=y

CONFIG_HEAP_MEM_POOL_SIZE=16384
CONFIG_MAIN_STACK_SIZE=4096

CONFIG_DEBUG_OPTIMIZATIONS=y
CONFIG_ASSERT=y

CONFIG_I2C=y

CONFIG_DISPLAY=y
CONFIG_DISPLAY_LOG_LEVEL_DBG=y
CONFIG_LOG_STRDUP_BUF_COUNT=16

CONFIG_SSD1306=y
CONFIG_SSD1306_SH1106_COMPATIBLE=n

CONFIG_LOG=y

# Console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=n
CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y
CONFIG_EARLY_CONSOLE=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y

# CFB
CONFIG_CFB_LOG_LEVEL_DBG=y
CONFIG_CHARACTER_FRAMEBUFFER=y
```

`display.overlay`:
```
&i2c0 {
    compatible = "nordic,nrf-twim";

    status = "okay";
    clock-frequency = <400000>;

    ssd1306@3c {
        compatible = "solomon,ssd1306fb";
        reg = <0x3c>;
        label = "SSD1306";
        width = <128>;
        height = <64>;
        segment-offset = <0>;
        page-offset = <0>;
        display-offset = <0>;
[... 357 lines cut ...]
 #else
-		SSD1306_SET_NORMAL_DISPLAY,
+            SSD1306_SET_NORMAL_DISPLAY,
 #endif
-	};
+    };
 
 #if DT_INST_NODE_HAS_PROP(0, reset_gpios)
 	struct ssd1306_data *driver = dev->data;
diff --git a/drivers/display/ssd1306_regs.h b/drivers/display/ssd1306_regs.h
index 2c4f628376..edeea1f5eb 100644
--- a/drivers/display/ssd1306_regs.h
+++ b/drivers/display/ssd1306_regs.h
@@ -12,6 +12,10 @@
 #define SSD1306_CONTROL_ALL_BYTES_CMD		0x00
 /* All following bytes will contain data */
 #define SSD1306_CONTROL_ALL_BYTES_DATA		0x40
+
+#define SSD1306_CONTROL_LAST_BYTE_CMD		0x00
+#define SSD1306_CONTROL_LAST_BYTE_DATA		0x40
+
 /* The next byte will contain a command */
 #define SSD1306_CONTROL_BYTE_CMD		0x80
 /* The next byte will contain data */
```

jeremyherbert, 2021-03-23:
Further digging implicates the `i2c_burst_write` function, the following change is all that is required to fix the issue. So it is the repeated start condition that is causing the problem; perhaps in the past for the nrf52840, `i2c_burst_write` did not generate a repeated start, and the regression is there?

```
diff --git a/drivers/display/ssd1306.c b/drivers/display/ssd1306.c
index 08b447e3b7..17dfa7d9c1 100644
--- a/drivers/display/ssd1306.c
+++ b/drivers/display/ssd1306.c
@@ -60,15 +60,16 @@ struct ssd1306_data {
 };
 
 #if DT_INST_ON_BUS(0, i2c)
+static uint8_t tmp_cmd[2048];
 static inline int ssd1306_write_bus(const struct device *dev,
                                    uint8_t *buf, size_t len, bool command)
 {
        struct ssd1306_data *driver = dev->data;
+       memcpy(tmp_cmd+1, buf, len);
+       tmp_cmd[0] = command ? SSD1306_CONTROL_ALL_BYTES_CMD :
+                                        SSD1306_CONTROL_ALL_BYTES_DATA;
 
-       return i2c_burst_write(driver->bus, DT_INST_REG_ADDR(0),
-                              command ? SSD1306_CONTROL_ALL_BYTES_CMD :
-                                        SSD1306_CONTROL_ALL_BYTES_DATA,
-                              buf, len);
+       return i2c_write(driver->bus, tmp_cmd, len+1, DT_INST_REG_ADDR(0));
 }
 
 #elif DT_INST_ON_BUS(0, spi)
```

jeremyherbert, 2021-03-23:
It appears that this problem has been encountered before on the Nordic devices: https://github.com/zephyrproject-rtos/zephyr/commit/e6fe54246509303e7081ce282f1d4eec2001b09a

jfischer-no, 2021-03-23:
There are display samples, `samples/subsys/display`, and different SSD1306 compatible shields, `boards/shields/ssd1306`. It would be more than appropriate to use these for reporting bugs, especially with an I2C display controller.

for `compatible = "nordic,nrf-twim";` you [need to add](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/bindings/i2c/nordic%2Cnrf-twim.yaml#L29) "zephyr,concat-buf-size " to your devicetree overlay:

```c
&arduino_i2c {                                                                                
        status = "okay";                                                                      
        zephyr,concat-buf-size = <2048>;                                                      
                                                                                              
        ssd1306@3c {                                                                          
                compatible = "solomon,ssd1306fb";                                             
                reg = <0x3c>;                                                                 
                label = "SSD1306";                                                            
                width = <128>;                                                                
                height = <64>;                                                                
                segment-offset = <0>;                                                         
                page-offset = <0>;                                                            
                display-offset = <0>;                                                         
                multiplex-ratio = <63>;                                                       
                segment-remap;                                                                
                com-invdir;                                                                   
                prechargep = <0x22>;                                                          
        };                                                                                    
}; 
``` 
But I think it is better to use `compatible = "nordic,nrf-twi";`. Anyway SSD1306 controller driver work fine:

[image]

closed 2021-03-23 by jfischer-no