zephyrproject-rtos/zephyr #33590: nrf: Debugging any test fails when CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_DWT is enabled

nordic-krch, 2021-03-23:
**Describe the bug**
Nordic board by default has null pointer detection using DWT enabled in all tests. As a result, any test fails in debug session ends in the assert during startup https://github.com/zephyrproject-rtos/zephyr/blob/76f35f29f01e57690576067f301ddf3df74e4815/arch/arm/include/aarch32/cortex_m/dwt.h#L151

**To Reproduce**
Steps to reproduce the behavior:
1. start debug session for any test on nordic SoC.

**Expected behavior**
Debug session of the test application should be ready without need to dig deep into very specific arch options.

**Impact**
Annoyance. User is wasting time to figure out what is wrong.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Commit SHA: 16d2ab790c

nordic-krch, 2021-03-23:
fyi @ioannisg

nashif, 2021-05-25:
closing as duplicate of #34222

closed 2021-05-25 by nashif