zephyrproject-rtos/zephyr #33593: acrn_ehl_crb: most of the test cases failed due to timeout

enjiamai, 2021-03-23:
**Describe the bug**
Most of the test cases failed due to timeout. The test cases run and being terminated before they completed.

**To Reproduce**
Steps to reproduce the behavior:

1. Make sure BIOS configuration is correct for ACRN, and include/arch/x86/memory.ld line 44, LOCORE_BASE modified from 0x1000 to 0x8000.
2. Run this command from the zephyrproject:
twister -p acrn_ehl_crb --device-testing --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11" -vv -T tests/kernel/
(you can pick one to run, for example:  -T tests/kernel/mutex/mutex_api to reproduce it. )

3. Check handler.log in twister.out in directory to see the error log.
You can see a line "Timed out while monitoring serial output on"

**Expected behavior**
The test case is passed as expected.

**Impact**
Serous impact. Most of the test cases failed,  and the pass rate down from 99.75% to 24.05% 

**Logs and console output**

```
2021-03-22 09:15:19,401 - twister - DEBUG - DEVICE: ACRN:\>vm_console 0
2021-03-22 09:15:19,402 - twister - DEBUG - DEVICE:
2021-03-22 09:15:19,402 - twister - DEBUG - DEVICE: ----- Entering VM 0 Shell -----
2021-03-22 09:15:19,480 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v2.5.99-ww11.3  ***
2021-03-22 09:15:19,480 - twister - DEBUG - DEVICE: Running test suite mutex_api
2021-03-22 09:15:19,480 - twister - DEBUG - DEVICE: ===================================================================
2021-03-22 09:15:19,481 - twister - DEBUG - DEVICE: START - test_mutex_lock_unlock
2021-03-22 09:15:19,481 - twister - DEBUG - DEVICE: PASS - test_mutex_lock_unlock in 0.0 seconds
2021-03-22 09:15:19,481 - twister - DEBUG - DEVICE: ===================================================================
2021-03-22 09:15:19,482 - twister - DEBUG - DEVICE: START - test_mutex_reent_lock_forever
2021-03-22 09:15:19,482 - twister - DEBUG - DEVICE: access resource from main thread
**2021-03-22 09:15:27,473 - twister - DEBUG - Timed out while monitoring serial output on acrn_ehl_crb**
2021-03-22 09:15:27,475 - twister - DEBUG - Process /opt/remotehw/acrn-test-pty.exp,ehlsku11 terminated outs: None errs None
2021-03-22 09:15:27,475 - twister - DEBUG - run status: acrn_ehl_crb/tests/kernel/mutex/mutex_api/kernel.mutex timeout
2021-03-22 09:15:27,477 - twister - ERROR - acrn_ehl_crb              tests/kernel/mutex/mutex_api/kernel.mutex          ^[[31mFAILED^[[39m: Timeout
2021-03-22 09:15:27,477 - twister - ERROR - see: ^[[33m/home/chenp1/work/zephyr-intel-ehl/zephyr/twister-out/acrn_ehl_crb/tests/kernel/mutex/mutex_api/kernel.mutex/handler.log^[[39m
```

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain: Zephyr SDK-0.12.3
 - Commit SHA or Version a6a4ed523b

**Additional context**
After git bisect,  we found it failed start from the commit: 915f4ac3c7584866d25f39b29366bbab8eea882b
"drivers/interrupt_controller: Get the IOAPIC RTEs dynamically"
And most of the test cases pass after revert it.

enjiamai, 2021-03-23:
@LeiW000 FYI

jenmwms, 2021-03-23:
@enjiamai The commit SHA a6a4ed523b does not exist in this repo, please update when you get a chance. It helps to have the bisect already, thanks!

enjiamai, 2021-03-23:
@jenmwms Thanks for your reminder, I update the commit information. It's commit SHA is a6a4ed523ba58c1e9850c62e423caef519934254 (Zephyr gitlab : tag: zephyr-v2.5.99-ww11.3)

By the way, I have already bisect it,  it started to failed from  915f4ac , thanks!

jenmwms, 2021-03-24:
I'm discussing with @tomasz about this. We are checking on priority of the loapic initialization call. I tried raising the priority but still see the issue. CONFIG_KERNEL_INIT_PRIORITY_DEFAULT=40 by default, so for debug changed to 50, 80, 1024 separately and no success:
```
diff --git a/drivers/interrupt_controller/intc_loapic.c b/drivers/interrupt_controller/intc_loapic.c
index 7fe2e4f849..e74f7ef735 100644
--- a/drivers/interrupt_controller/intc_loapic.c
+++ b/drivers/interrupt_controller/intc_loapic.c
@@ -428,7 +428,7 @@ static int loapic_device_ctrl(const struct device *port,
 SYS_DEVICE_DEFINE("loapic", loapic_init, loapic_device_ctrl, PRE_KERNEL_1,
                  CONFIG_KERNEL_INIT_PRIORITY_DEFAULT);
 #else
-SYS_INIT(loapic_init, PRE_KERNEL_1, CONFIG_KERNEL_INIT_PRIORITY_DEFAULT);
+SYS_INIT(loapic_init, PRE_KERNEL_1, 1024);
 #endif   /* CONFIG_PM_DEVICE */
```

jenmwms, 2021-03-24:
There is only 1 difference in the .config before/after the bisect, and it is indeed the kconfig which was removed in that patch to allow for dynamic IOAPIC RTEs instead of fixed number of 24 RTEs from legacy days.
```
CONFIG_IOAPIC_NUM_RTES=24
```

jenmwms, 2021-03-24:
> I tried raising the priority but still see the issue.

In that comment I mentioned testing `SYS_INIT(loapic_init, PRE_KERNEL_1, <value greater than 40>)`, but this parameter needs to be lower number to have higher priority ranking. So I tried 5 and 39, but still see the issue. (For info on parameter see https://docs.zephyrproject.org/latest/reference/drivers/index.html#initialization-levels).

So it seems the issue is not due to priority of the loapic init. Now looking for a different potential root cause.

jenmwms, 2021-03-24:
Hi @enjiamai In the bug description you mention most of the tests do not actually fail but it is a timeout with twister. Does this mean that you were successful to manually test with west build single test and flash the board, then use simple script to step through VM console? I will learn this manual process too. So far I have used twister mostly.

jenmwms, 2021-03-24:
Some of the #define macros were moved slightly in that bisected patch, which may have introduced new dependency. I'll try walking it back some, and I'm reaching out to learn more and see if this was a needed/intentional change.

enjiamai, 2021-03-25:
> Hi @enjiamai In the bug description you mention most of the tests do not actually fail but it is a timeout with twister. Does this mean that you were successful to manually test with west build [...]

Hi @jenmwms Yes, it runs successful in west, but I guess this failure is not due to twister.  

And the procedure of run by west is: 
1. build singe test:
```
west build -p auto -b acrn_ehl_crb tests/kernel/mutex/mutex_api
```
2. open another terminal to capture the message of the serial output, for example:
```
sudo picocom -b 115200 /dev/ttyUSB0
```
3. flash the board:
```
west flash /opt/remotehw/remotehw-x86-acrn.sh ehlsku11
```
4. watch the output (result) in the terminal.

jenmwms, 2021-03-26:
Note this issue persists also when using an updated version of ACRN (v2.3).

andyross, 2021-03-29:
Just blindly guessing: how likely is it that ACRN is simply lying about the number of RTEs, or has a zero in that field or whatever?  Does Linux under ACRN even use the IO-APIC?  It's possible that this is just an incomplete emulation and we're the first to stumble on it.

jenmwms, 2021-03-30:
Using printk to debug a little, the `ioapic_rtes` is getting set to 48 based on the __IoApicGet( ) implemented in the patch at 915f4ac. Hardcoding to `ioapic_rtes=24` instead of snippet below allows the test to proceed and pass. Not sure what the real fix is yet though.
```
ioapic_rtes = ((__IoApicGet(IOAPIC_VERS) &
			(IOAPIC_MRE_MASK) >> IOAPIC_MRE_POS)) + 1;
```

jenmwms, 2021-03-30:
> Just blindly guessing: how likely is it that ACRN is simply lying about the number of RTEs, or has a zero in that field or whatever?

Hi @andyross, it's possible. I reached out, not sure yet. I don't have a way to debug ACRN side right now. However, I observed the value retrieved from that register at both the 'working' 24 and 'not working' 48. It remained unchanged, too.

Looking at https://github.com/zephyrproject-rtos/zephyr/blob/ae4a28ca08ee69dba72f0d93cc2ca311d41bf885/drivers/interrupt_controller/intc_ioapic.c#L140 and pulling out out the MRE read,

 `uint32_t debugIoApicGet = __IoApicGet(IOAPIC_VERS);` 

gives us 

`ioapic_rtes 24 debugIoApicGet 3080209`

jenmwms, 2021-03-31:
I'm trying a debug exercise Andy and I discussed to use different ioapic_rtes (24, 31, 32, ..., 48), and the purpose is to find which (suspect a single) interrupt is not working properly (slows down our tests). Well, I don't find any other that works well (tests execute at expected rate), only 24. BUT - *37* behaves differently than any other so far...Zephyr does not boot. Perhaps it is reserved for ACRN. Looking into it

jenmwms, 2021-03-31:
Filed a bug about this on ACRN github https://github.com/projectacrn/acrn-hypervisor/issues/5897

jsun26intel, 2021-04-02:
I just find there is a hard coded CONFIG_APIC_TIMER_IRQ=24, I am not familiar with how the IRQ routing in Zephyr, but when IOAPIC RTE/pins increase, it obviously overlap with IOAPIC IRQ.

enjiamai, 2021-04-02:
@jenmwms @jsun26intel  After validating, all the test case **pass** except the original failed one(arch.interrupt),  by following change:  
```
diff --git a/boards/x86/acrn/acrn_ehl_crb_defconfig b/boards/x86/acrn/acrn_ehl_crb_defconfig
index 932a27588e..e2cb989875 100644
--- a/boards/x86/acrn/acrn_ehl_crb_defconfig
+++ b/boards/x86/acrn/acrn_ehl_crb_defconfig
@@ -14,3 +14,4 @@ CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=1900000000
 CONFIG_BUILD_OUTPUT_BIN=y
 CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=n
 CONFIG_KERNEL_VM_SIZE=0x1000000
+CONFIG_APIC_TIMER_IRQ=48 
```

jenmwms, 2021-04-05:
> I just find there is a hard coded CONFIG_APIC_TIMER_IRQ=24, I am not familiar with how the IRQ routing in Zephyr, but when IOAPIC RTE/pins increase, it obviously overlap with IOAPIC IRQ.

Thanks! this value of 24 is a default for several of the x86 SoCs like [apollo_lake](https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/x86/apollo_lake/Kconfig.defconfig). However I don't see it specified for the ia32 (which is 64 bit, despite the naming) so it must be coming from another level of kconfig if not set by board or SoC. Looks like we need to add this for acrn_ehl_crb board level.

jenmwms, 2021-04-05:
> +CONFIG_APIC_TIMER_IRQ=48

Thanks @enjiamai ! Can you please offer some insight why this needs to differ from the default which is used widely across other x86 boars and SoCs (which use default 24). So far, many configurations from ACRN on EHL CRB align with the native HW EHL CRB, which makes me curious to understand if this deviation from native is a workaround here or resolving the real issue at hand.

jenmwms, 2021-04-05:
I'll submit a patch with this change so we can discuss :)

closed 2021-04-06 by nashif

jsun26intel, 2021-04-09:
please double check zephyr's implementation.
EHL native doesn't use APIC timer, it use HPET timer by default, whereas ACRN has no vHPET implementation, so we have to use APIC timer instead. That's why the issue could not be reproduced on EHL native. Maybe you can have a try on APIC timer with SMP disabled.

On Zephry's apic timer implementation (drivers/timer/apic_timer.c), it would enable the hardcoded APIC IRQ(24), thus the ioapic pin would be enabled by below function:

void arch_irq_enable(unsigned int irq)
{
    if (IS_IOAPIC_IRQ(irq)) {
        z_ioapic_irq_enable(irq);
    } else {
        z_loapic_irq_enable(irq - z_loapic_irq_base());
    }
}

PR #34018 (2021-04-06): boards: x86: acrn_ehl_crb: fix config Apic timer IRQ value
The default APIC_TIMER_IRQ of 24 causes significant slowdown
in general tests and samples execution on this board. As the
value of ioapic RTEs increases, e.g. 48 in ACRN EHL CRB, the
APIC TIMER IRQ is stepped on. This commit tunes the config
for APIC_TIMER_IRQ to 48 for this board.

Fixes #33593
commit 3c563648 (2021-04-06): boards: x86: acrn_ehl_crb: fix config Apic timer IRQ value

The default APIC_TIMER_IRQ of 24 causes significant slowdown
in general tests and samples execution on this board. As the
value of ioapic RTEs increases, e.g. 48 in ACRN EHL CRB, the
APIC TIMER IRQ is stepped on. This commit tunes the config
for APIC_TIMER_IRQ to 48 for this board.

Fixes #33593