zephyrproject-rtos/zephyr #33609: Question about memory usage of the binary zephyr.exe

nqduy35, 2021-03-23:
I'm new with Zephyr OS, so my question is very basic. I built a **hello_world** program for the **native_posix** board. I want to know the RAM usage of my program, then I tried to use the two following approaches: 
(1) Running `$ west build -t ram_report ` as presented in [here](https://docs.zephyrproject.org/latest/guides/optimizations/tools.html#footprint-and-memory-usage). The result is 5242 B (~5 KB).
(2) Running `$ west build -t run `, then uses the **ps** tool to check the RES (physical memory consuming) with the command `$ ps aux | grep zephyr.exe`. The result is 1104 KB (~1.1 MB). 

How should I understand these two results. Thank you in advance.

aescolar, 2021-03-24:
`ram_report` for posix arch boards (including native_posix) does not do anything (of any use).
When you build targetting native_posix, your code is built linked to several host shared libraries (including the C library), so the total memory usage reported by `ps` or other similar tools is going to be quite big (when accounting for the total of those shared libraries in memory).
Given that when you build for native_posix, you build a normal linux program, you can use normal linux tools to check its memory usage. The results will be "correct", but the output may not be too useful depending on what you are trying to do.

nqduy35, 2021-03-24:
Thank @aescolar for the answer. Regarding your answer, when I want to build my **hello_world** program to another board NOT posix arch (e.g. [nrf52 DK](https://docs.zephyrproject.org/latest/boards/arm/nrf52dk_nrf52832/doc/index.html)), then I can use `ram_report` to calculate the actual RAM required by the program as it is run on the actual device? In fact, I don't have the device at this moment, but I need to calculate the RAM footprint.

aescolar, 2021-03-24:
> when I want to build my hello_world program to another board NOT posix arch (e.g. nrf52 DK), then I can use ram_report to calculate the actual RAM required by the program as it is run on the physical device? 

yes, you just need to build for the actual target you will use

nqduy35, 2021-03-24:
It is clear to me now. Many thanks to @aescolar and @carlescufi. I close the issue.

closed 2021-03-24 by nqduy35