zephyrproject-rtos/zephyr #33627: Provide alternative nvs_init that will take const struct *device instead of device name

de-nordic, 2021-03-23:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nvlsianpu, 2021-03-23:
I don't have strong preference. We can introduce nvs_init_with_dev() or something like that.

Laczen, 2021-03-24:
Maybe there is no need to make a new function. If nvs_init() is called with `fs.flash_device` already defined the call to `get_device_binding` could be skipped.

nvlsianpu, 2021-03-25:
@Laczen I think this would be better to have API extended or modified.
What you have suggested will works, but will be looking as a hack:

``` C
fs.flash_device = flash_area_get_device(&fa);
rc = nvs_init(&fs, NULL);
```

I would either change nvs_init()
``` C
int nvs_init(struct nvs_fs *fs)
```
or add new API: 
``` C
int nvs_init_pure(struct nvs_fs *fs)
```

Are you are reluctant to API update?

Laczen, 2021-03-25:
@nvlsianpu, I am not reluctant to API update but would make the transition as effortless as possible:
a. We keep `nvs_init()` as is and give a warning when it is not called with *dev_name=NULL a warning is given that it should be updated.
b. After some time a switch is made to define as `nvs_init(struct nvs_fs *fs)`

Instead of using the `flash_area` api I would just use `fs.flash_device = DEVICE_DT_GET(...)`

martinjaeger, 2022-09-23:
Looks like this has been implemented in #43260 by @gmarull. Please re-open if you disagree.

closed 2022-09-23 by martinjaeger