zephyrproject-rtos/zephyr #33633: NXP imx rt1064 evk: Application does not boot when flash/flexSPI driver is enabled

hendriklehmkuhl, 2021-03-23:
**Describe the bug**
I am currently trying to build and flash the MCUBoot bootloader to a nxp mimxrt1064 evaluation board. Building and flashing using west works fine without any warnings or error messages but unfortunately the application does not boot. There is nothing on the console and also GDB debugging using a Segger J-Link does not work. 

After searching for the reason for a while I noticed that the same behaviour occurs when I try to include the flash driver in other sample applications as well. So I am able to provide a minimal example to reproduce this issue:

**To Reproduce**
Steps to reproduce the behavior:
1. Add CONFIG_FLASH=y to samples/hello_world/prj.conf
2. Build the hello world application (no warnings or errors should show up)
3. Flash the application to the mimxrt1064_evk
4. See that the application does not boot and nothing is shown on the console

**Expected behavior**
Show the zephyr boot banner and boot as usual when including the flash driver.

**Impact**
I am not able to integrate MCUBoot with Zephyr on my RT1064 evaluation board.

**Environment :**
 - OS: Windows
 - Toolchain: gnu arm embedded
 - Commit: Current master (2.5.0-rc1)

pdgendt, 2021-03-26:
Zephyr currently doesn't support `mcuboot` for the i.MX RT1064 internal QSPI flash.

I've submitted [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/33701) that should help you get started.

pdgendt, 2021-05-05:
Hey @hendriklehmkuhl, can you verify again, now that [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/34845) has been merged?

As mentioned, check that you build the bootloader with `CONFIG_BOOT_MAX_IMG_SECTORS=512`.

MaureenHelm, 2021-05-12:
Fixed by #33701 and #34845. Please reopen if you still the issue after pulling in these changes.

closed 2021-05-12 by MaureenHelm