zephyrproject-rtos/zephyr #33635: subsys/ipc/openamp sample on QEMU not working when debugging

MartinSchoenstedt, 2021-03-23:
**Describe the bug**
Trying to understand the openamp sample I wanted to debug it via QEMU. However, the sample stops working when adding the `-S` option

**To Reproduce**
Steps to reproduce the behavior:
1. go to `zephyrproject/zephyr/samples/subsys/ipc/openamp`
2. build with `west build -p -b mps2_an521 .` (adding `-t run` does not work here)
3. run the sample with `qemu-system-arm` as `build/build.ninja` describes in line 882
4. Sample works without problems
5. repeat command but add options `-s -S` to the `qemu-system-arm` call for debugging
6. start gdb-server and continue
7. Remote core does not start

**Environment:**
 - OS: Linux 64-bit
 - Zephyr version: 2.5.99

galak, 2021-03-23:
@MartinSchoenstedt I'm guessing this isn't a bug, but more how qemu works w/multicore.

MartinSchoenstedt, 2021-03-24:
yes, that could be the case. I was wondering if the problem has occured to anyone else before or if I am the first trying to debug multicore qemu applications.

closed 2021-03-24 by MartinSchoenstedt

galak, 2021-03-24:
@MartinSchoenstedt 

Take a look at:

https://lists.nongnu.org/archive/html/qemu-devel/2020-01/msg07035.html

If I use the gdb commands listed here things seem to work for me.