zephyrproject-rtos/zephyr #33641: API Meeting Minutes [Test]

nashif, 2021-03-23:
Minutes from the API meeting...

nashif, 2021-03-23:
https://github.com/zephyrproject-rtos/zephyr/issues/23727#issuecomment-805051697

carlescufi, 2021-03-30:
2021.03.30:

https://github.com/zephyrproject-rtos/zephyr/pull/33566#issuecomment-810396251
https://github.com/zephyrproject-rtos/zephyr/pull/33602#issuecomment-810406928
Started discussion about pinctrl again: https://github.com/zephyrproject-rtos/zephyr/issues/22748#issuecomment-717420438

carlescufi, 2021-04-06:
2021.06.04:

- Dictionary-based logging: https://github.com/zephyrproject-rtos/zephyr/pull/30539#issuecomment-814274278
- Doxygen warnings and `#ifdef __DOXYGEN__`: https://github.com/zephyrproject-rtos/zephyr/pull/33972#issuecomment-814274671

mbolivar-nordic, 2021-04-06:
Why are we doing this here? This is inconsistent with other meetings that track minutes in a Google doc and may confuse external observers.

carlescufi, 2021-04-13:
For consistency with the rest of the meetings, closing this in favor of:
https://docs.google.com/document/d/1lv-8B5QE2m4FjBcvfqAXFIgQfW5oz6306zJ7GIZIWCk/edit#

closed 2021-04-13 by carlescufi

nashif, 2021-04-27:
> Why are we doing this here? This is inconsistent with other meetings that track minutes in a Google doc and may confuse external observers.

yeah, that was the original ask, have minutes in general, but then the idea of google docs for the API meeting was not appealing for some... :)

mbolivar-nordic, 2021-04-27:
> > 
> 
> yeah, that was the original ask, have minutes in general, but then the idea of google docs for the API meeting was not appealing for some... :)

I can totally sympathize with that; I write minutes in my editor and then copy them into Gdocs later. I just think we should be consistent until we can all agree on an alternative.