zephyrproject-rtos/zephyr #33645: Random MAC after RESET - NRF52832

tcpipchip, 2021-03-23:
**Describe the bug**
I implemented a Beacon using NRF52832 to broadcast the BME680 data!
All the time that i reset the NRF52832, a different MAC is detected my  BLE SCANNER.

**To Reproduce**
Reseting the ZEPHYR, the BLE SCANNER detects different MAC.
**Expected behavior**
Allways the Same MAC

**Impact**
I had to use the CHIP ID to identify my ZEPHYR module and broadcast it!

Thanks!

joerchan, 2021-03-31:
@tcpipchip It looks like you have privacy disabled `CONFIG_BT_PRIVACY=n` and have a non-connectable advertiser.
In this case the advertiser will use a new NRPA address each time it starts to advertise. If this behavior is not wanted you can use
BT_LE_ADV_OPT_USE_IDENTITY to advertise with the identity address.

tcpipchip, 2021-03-31:
@joerchan 
Thanks for the feedback!

I have other question
sprintf(g_localName,"%s%x","INDRIYA-G-",NRF_FICR->DEVICEADDR[0]); 

/* Set Scan Response data */
	static struct bt_data sd[] = {
	BT_DATA(BT_DATA_NAME_COMPLETE, g_localName, 17),
	};

Why sometimes advertsing the INDRIYA-G-XXXXXX and sometimes not adverstising the INDRIYA-G-XXXXXX ?

Thanks again!

joerchan, 2021-04-06:
@tcpipchip Not much information to go on there, are you sure it isn't just the difference in the advertising PDU? You have put the name in the scan-response, so you won't see it in the ADV_IND pdu.

tcpipchip, 2021-04-06:
@joerchan 
I want to broadcast the localname....
What do you suggest

joerchan, 2021-04-06:
Putting the localname in the scan-response achieves that, but the scanner needs to be an active one.
The other option is to put it in the advertising data (ADV_IND).

tcpipchip, 2021-04-06:
i had choosen the second option! It[s working
Have you a little example showing the scan-response example

joerchan, 2021-04-06:
https://docs.zephyrproject.org/2.5.0/reference/bluetooth/gap.html?highlight=adv_start#c.bt_le_adv_start
if you put it as `ad` in will be in the advertising data `ADV_IND`, if you put is `sd` it will be in the scan response data `SCAN_RSP`.

tcpipchip, 2021-04-06:
Yes, i did that

Maybe is the BLE SCANNER (ARDUINO) ESP32 example bug :(

joerchan, 2021-04-06:
Maybe, can you verify with your phone that the name shows up? You can use the  nRF Connect app

tcpipchip, 2021-04-06:
they show!

But, is using ESP32 as ble scan :(

joerchan, 2021-05-19:
Closing as original question and follow-up question has been answered.

closed 2021-05-19 by joerchan