zephyrproject-rtos/zephyr #33646: Expose net_ipv4_create, net_ipv6_create, and net_udp_create in standard header

galak, 2021-03-23:
Currently some of the modem drivers (drivers/modem/hl7800.c and drivers/modem/wncm14a2a.c) are calling `net_ipv6_create`, `net_ipv4_create` and `net_udp_create` which requires having:

`zephyr_library_include_directories(${ZEPHYR_BASE}/subsys/net/ip)` in `CMakeLists.txt`.  If these are suppose to be APIs that can be called on the networking subystem we should look to move them into `include/net/` somewhere.

galak, 2021-03-23:
@jukkar was looking at various places we do `zephyr_library_include_directories` and if there is some implied cleanup that needs to happen.  This is one area related to networking interfaces and modem.  Don't really know much about it beyond that.

jukkar, 2021-03-24:
I would not want to create new APIs for these, as they are meant for internal use. Is there some issue if `zephyr_library_include_directories` is used in the driver?

galak, 2021-03-24:
> I would like not to expose these APIs by putting them to `include/net`, as they are meant for internal use. Is there some issue if `zephyr_library_include_directories` is used in the driver?
> Calling these private APIs from drivers is ok, but I would like to avoid exposing them to applications.

There's not a specific issue, just trying to ensure some level of encapsulation - even if its between a subsystem a driver.

galak, 2021-03-24:
@jukkar will close this out, we need a general solution/agreement on how subsystems export APIs to other parts of zephyr (not application).

closed 2021-03-24 by galak