zephyrproject-rtos/zephyr #33655: Add support for board: Nucleo-L412RB

GunZi200, 2021-03-24:
I bought this board to try out the L4 series and noticed L412RB is not supported: https://docs.zephyrproject.org/latest/boards/index.html

I'm thinking about adding this support to Zephyr myself as a challenge/exercise.

* https://www.digikey.com/en/products/detail/stmicroelectronics/NUCLEO-L412RB-P/9656230

GunZi200, 2021-03-24:
I've got the hello world sample working 😃 
[image]

GunZi200, 2021-03-25:
@erwango I've hit a wall with PWM. Seems this board's User LED uses a "**Complementary PWM**" output.. 

  * Pin is PB13
  * TIM1_CH1N.

From what I gather from Googling is that this is an advanced feature which must be activated specifically.

Is this type of PWM not supported in Zephyr? I can't find any specific information that says to treat it differently. Though maybe I'm not looking in the right places.

This is what my board's device tree looks like at the moment. I see no LED action. The examples "Blinky PWM" and "Fade LED" show no errors when I run them on the board.

Any ideas? Should I perhaps skip PWM for initial support?

```c
#include <st/l4/stm32l412rbtx-pinctrl.dtsi>
#include <st/l4/stm32l412XB.dtsi>

/ {
	model = "STMicroelectronics STM32L412RB-NUCLEO board";
	compatible = "st,stm32l412rb-nucleo";

	chosen {
		zephyr,console = &usart2;
		zephyr,shell-uart = &usart2;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
	};

	leds {
		compatible = "gpio-leds";
		green_led: led_0 {
			gpios = <&gpiob 13 GPIO_ACTIVE_HIGH>;
			label = "User LD3";
		};
	};

	pwmleds {
		compatible = "pwm-leds";
		green_pwm_led: green_pwm_led {
			pwms = <&pwm2 1 0 PWM_POLARITY_INVERTED>;
		};
	};

	aliases {
		led0 = &green_led;
		pwm-led0 = &green_pwm_led;
	};
};

&timers1 {
	status = "okay";

	pwm2: pwm {
		status = "okay";
		pinctrl-0 = <&tim1_ch1n_pb13>;
	};
};
```

erwango, 2021-03-26:
@GunZi200 

When looking to ref manual for this board [UM1956.pdf](https://www.st.com/resource/en/user_manual/dm00231744-stm32-nucleo32-boards-mb1180-stmicroelectronics.pdf), user LED is indicated as PB3, and I see you're setting it as PB13. Can you cross check ?
This would mean using `timer2s` and `tim2_ch2_pb3`.

GunZi200, 2021-03-26:
> @GunZi200
>
> When looking to ref manual for this board [UM1956.pdf](https://www.st.com/resource/en/user_manual/dm00231744-stm32-nucleo32-boards-mb1180-stmicroelectronics.pdf), user LED is indicated as [...]

No I'm sure it is PB13 since the Blinky example (without PWM) works. [UM2206.pdf](https://github.com/zephyrproject-rtos/zephyr/files/6212037/dm00387966-stm32-nucleo64p-boards-mb1319-stmicroelectronics.pdf) I also looked at the schematic drawing to confirm this.

[image]

erwango, 2021-03-26:
> No I'm sure it is PB13 since the Blinky example (without PWM) works. UM2206.pdf I also looked at the schematic drawing to confirm this.

Right, mixed with nucleo_l412**k**b which is a nucleo-32. So let me check and come back to you.

closed 2021-04-06 by carlescufi