zephyrproject-rtos/zephyr #33667: tests: kernel: timer: Test timeout_abs from kernel/timer/timeout_abs hangs causing test scenarios to fail

PerMac, 2021-03-24:
**Describe the bug**
The test timeout_abs from kernel.timer.tickless and kernel.timer scenarios at kernel/timer/timeout_abs hangs on nrf5340dk_nrf5340_cpuappns causing this scenarios to fail (test sleep_abs is not executed)

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk connected
2. go to your zephyr dir
3. run `./scripts/twister -T tests/kernel/timer/timer_api/ -p nrf5340dk_nrf5340_cpuappns --device-testing --device-serial /dev/ttyACM2 --jobs 1 -v -v --inline-logs`
4. See error

**Expected behavior**
Test passes

**Impact**
Not clear

**Logs and console output**
```
...
START - test_timer_remaining
PASS - test_timer_remaining in 0.51 seconds
===================================================================
START - test_timeout_abs
```
and hangs. Then is terminated with a timeout

**Environment (please complete the following information):**
 - OS:  Ubuntu 18.04
 - Toolchain zephyr sdk 0.12.2
 - Commit SHA or Version used zephyr-v2.5.0-1463-gc59cf

**Additional context**
This is a different issue than #32839. There the test fails due to an assertion error. Here it hangs.

ioannisg, 2021-03-24:
@anangl I wonder if you could take a look at this bug.

ioannisg, 2021-04-15:
This seems to fail also for nrf9160dk_nrf9160ns, in the same way (@PerMac confirm?)

ioannisg, 2021-04-15:
@anangl did you have a chance to pick this bug investigation?

anangl, 2021-04-15:
> @anangl did you have a chance to pick this bug investigation?

The problem is that this loop: https://github.com/zephyrproject-rtos/zephyr/blob/28a730901724ff71d8d1b5eb7cc73d30e132c539/tests/kernel/timer/timer_api/src/main.c#L757-L761 has no chance to finish because the execution of `rem_ticks = k_timer_remaining_ticks(&remain_timer);` always takes more than one tick (for the non-secure image; for the secure one, it is a bit shorter, so the test passes).

carlescufi, 2021-04-20:
@andyross what do you suggest we do for configurations where the execution of the loop will always exceed one tick given that the calls are slow to execute when we are in TEE non-secure mode?

andyross, 2021-04-22:
Just noticing this comment.  What's the reason that the remaining check is taking longer than a tick?  That call is just a wrapper around z_timeout_remaining(), which is a pure kernel thing that just queries state.  The only interaction with the timer driver is a call to sys_clock_elapsed() to get the current ticks since the last interrupt.

The intent is that sys_clock_elapsed() should be implementable with a single register read, it should require any blocking on hardware state.  If it actually requires waiting for a full 30+ us tick, that.... actually a pretty severe performance bug.  The kernel queries this function **a lot** and assumes it's fast.

(FWIW: there's an Intel-internal platform with a similar glitch right now where it needs to wait for that kind of period on register **writes** only, and that's causing a bunch of problems too.  Doing it on a read seems like it's going to be a non-starter for any kind of performance-senstive work.)

Broadly: what is this trying to achieve?  And if this is unavoidable, maybe this configuration really just doesn't want to be doing tickless and would be happy with 1024 Hz ticks or something (the kernel only needs to constantly check the current time with tickless, obviously).

PR #35510 (2021-05-26): manifest: update TF-M with latest upstream
The TF-M update brings in the latest PSA crypto headers,
after TF-M switched formally to MBedTLS 2.26.0. It also
brings in some important fixes for cache enabling in nRF
platforms.

Fixes #33423.
Fixes #33667.
Fixes #34962 

Supersedes #35455
closed 2021-05-26 by galak