zephyrproject-rtos/zephyr #33688: hci_usb example locks up on connect attempt

abferm, 2021-03-24:
**Describe the bug**
The hci_usb example times out on connect, and will not respond to further commands.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nrf52840dk_nrf52840 samples/bluetooth/hci_usb
2. west flash
3. Switch to nrf usb interface
4. sudo hcitool -i hci1 lescan
5. sudo hcitool -i hci1 lecc <addr>
6. See "Connection timed out"
7. sudo hcitool -i hci1 lescan
8. See "Input/output error"
9. Press reset button
10. Scan works again

**Expected behavior**
Connect to BLE device!

**Impact**
This is a showstopper! I am looking to use an nrf based usb dongle to leverage newer bluetooth features, and wanted to use the device as an hci device in the short term, and later develop a more complex firmware specific to my application.

**Logs and console output**
```
aferm@alex-desktop:~$ sudo hcitool -i hci1 lescan
LE Scan ...
20:9C:0C:F1:86:F3 (unknown)
00:00:09:B0:04:2B (unknown)
47:68:04:6C:61:79 (unknown)
47:68:04:6C:61:79 (unknown)
49:7A:ED:F9:7E:77 (unknown)
49:7A:ED:F9:7E:77 (unknown)
20:CD:39:A0:AB:EE (unknown)
20:CD:39:A0:AB:EE Mooshimeter V.1
4C:87:5D:24:B1:31 (unknown)
4C:87:5D:1A:39:BA (unknown)
4C:87:5D:24:B1:31 LE-reserved_T
4C:87:5D:1A:39:BA LE-Sparkle Motion
CC:C0:9E:95:98:60 (unknown)
CC:C0:9E:95:98:60 (unknown)
CB:52:5D:22:40:AB Oil_Drop
CB:52:5D:22:40:AB (unknown)
C4:11:0D:44:2A:E3 Tile
36:6A:A2:63:89:EF (unknown)
5C:88:DF:18:FC:0D (unknown)
5C:88:DF:18:FC:0D (unknown)
C4:11:0D:44:2A:E3 (unknown)
^Caferm@alex-desktop:~$ sudo hcitool -i hci1 lecc 20:CD:39:A0:AB:EE
Could not create connection: Connection timed out
aferm@alex-desktop:~$ sudo hcitool -i hci1 lescan
Set scan parameters failed: Input/output error

```

**Environment (please complete the following information):**
 - OS: `Linux alex-desktop 5.8.0-45-generic #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
 - BlueZ: 5.53-0ubuntu3
 - Toolchain zephyr-sdk-0.12.3
 - Installed 0.12.3 by following [these directions](https://docs.zephyrproject.org/latest/getting_started/index.html)

carlescufi, 2021-03-24:
1. Are you using master?
2. Can you please capture a trace with `btmon`?

abferm, 2021-03-24:
I used `west init ~/zephyrproject`, and `west update`, this appears to use master. I will get an exact commit and capture a trace when I get back to my workstation in the morning.

abferm, 2021-03-25:
I built the firmware used above from commit `f9a0e20432818308bead889fece60869337a32cb`
To me this looks like btmon is reporting success for the connection create, but hcitool is not. I also don't understand why scanning would be "disallowed".

I have also tried this with my custom application, which is based on go-ble. It works fine with other hci devices, but hangs when trying to connect using the hci_usb example.

Here is console output:
```
aferm@alex-desktop:~/zephyrproject/zephyr$ sudo hcitool -i hci1 lescan
LE Scan ...
46:61:2C:E1:8B:F4 (unknown)
46:61:2C:E1:8B:F4 (unknown)
4F:B7:48:46:D0:84 (unknown)
58:12:4A:0D:CE:B9 (unknown)
58:12:4A:0D:CE:B9 (unknown)
CC:C0:9E:95:98:60 (unknown)
C4:11:0D:44:2A:E3 Tile
C4:11:0D:44:2A:E3 (unknown)
CB:52:5D:22:40:AB Oil_Drop
CB:52:5D:22:40:AB (unknown)
0F:C3:F7:0E:6E:EC (unknown)
CC:C0:9E:95:98:60 (unknown)
4F:B7:48:46:D0:84 (unknown)
52:5C:CF:65:DA:2B (unknown)
77:55:3D:BF:F2:19 (unknown)
77:55:3D:BF:F2:19 (unknown)
6A:DC:6D:4B:8B:DC (unknown)
6A:DC:6D:4B:8B:DC (unknown)
^Caferm@alex-desktop:~/zephyrproject/zephyr$ sudo hcitool -i hci1 lecc C4:11:0D:44:2A:E3
Could not create connection: Connection timed out
aferm@alex-desktop:~/zephyrproject/zephyr$ sudo hcitool -i hci1 lescan
Set scan parameters failed: Input/output error
```

Here is btmon output:
```
Bluetooth monitor ver 5.53
= Note: Linux version 5.8.0-45-generic (x86_64)                                                                                                    0.979830
= Note: Bluetooth subsystem version 2.22                                                                                                           0.979832
= New Index: 00:00:00:00:00:00 (Primary,USB,hci1)                                                                                                  0.979834
= Open Index: 00:00:00:00:00:00                                                                                                                    0.979834
= Index Info: 00:00:00:00:00:00 (The Linux Foundation)                                                                                             0.979835
= New Index: 9C:D2:1E:D4:5D:90 (Primary,USB,hci0)                                                                                                  0.979835
= Open Index: 9C:D2:1E:D4:5D:90                                                                                                                    0.979836
= Index Info: 9C:D2:1E:D4:5D:90 (Atheros Communications, Inc.)                                                                                     0.979836
@ MGMT Open: bluetoothd (privileged) version 1.17                                                                                         {0x0001} 0.979837
@ RAW Open: hcitool (privileged) version 2.22                                                                                             {0x0003} 5.460537
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                      #1 5.460577
        Type: Active (0x01)
        Interval: 10.000 msec (0x0010)
        Window: 10.000 msec (0x0010)
        Own address type: Public (0x00)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                     #2 5.461969
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                          #3 5.462061
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                     #4 5.462929
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 43                                                                                                       #5 5.486921
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: 46:61:2C:E1:8B:F4 (Resolvable)
        Data length: 31
        Flags: 0x1a
          LE General Discoverable Mode
[... 220 lines cut ...]
        Peer address type: Public (0x00)
        Peer address: C4:11:0D:44:2A:E3 (OUI C4-11-0D)
        Own address type: Public (0x00)
        Min connection interval: 18.75 msec (0x000f)
        Max connection interval: 18.75 msec (0x000f)
        Connection latency: 0 (0x0000)
        Supervision timeout: 32000 msec (0x0c80)
        Min connection length: 0.625 msec (0x0001)
        Max connection length: 0.625 msec (0x0001)
> HCI Event: Command Status (0x0f) plen 4                                                                                                     #26 42.421894
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                                                                                                                     {0x0003} 67.424579
@ RAW Open: hcitool (privileged) version 2.22                                                                                            {0x0003} 85.803364
[... cut ...]
        Own address type: Public (0x00)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                   #28 85.804862
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Command Disallowed (0x0c)
@ RAW Close: hcitool                                                                                                                     {0x0003} 85.805085
```

abferm, 2021-03-25:
I see the same behavior at `zephyr-v2.5.0` and `1804d7f5f07357fd9ab976621d195be36ba8a848`

carlescufi, 2021-03-25:
Thanks for the logs.
Here's what's going on. The Host issues an `LE Create Connection` command, to which the controller responds with `Command Status`. Without waiting for the outcome of that connection, and without the Host (BlueZ) cancelling the connection creation, you try to enable scanning to find more devices, by setting the parameters first. But the controller replies with `Command Disallowed`, because a connection creation attempt is in progress and that means the scanner is busy. This is likely a limitation of our controller, that cannot both scan and create a connection simultaneously, and BlueZ doesn't handle it well. What it should do is to send an `LE Create Connection Cancel` command, and then start scanning.

@cvinayak can you confirm?

abferm, 2021-03-25:
As shown in the terminal output above, hcitool times out waiting for the connection to be created. I do not see a way to cancel the connection create call using hcitool. The closest option is `ledc` (disconnect), which gives the same output as scan.

carlescufi, 2021-03-25:
Right, but the timeout you mention is the host’s, the controller is still trying to connect.

abferm, 2021-03-25:
I understand this, however this condition remains until the hardware is reset. Shouldn't there be some point at which the controller determines the connection has failed?

I have not been able to successfully establish any connections using this sample.

carlescufi, 2021-03-25:
No, there is no timeout in the controller. It will keep scanning forever until it establishes a connection or receives a connection creation cancellation command

carlescufi, 2021-03-26:
@abferm this is a BlueZ bug, because the controller clearly states that it doesn't support scanning while creating a connection, and it exposes that information to the Host as part of the `LE Read Supported States` command.

sjanc, 2021-03-26:
hcitool is old low level tool which is basically sending raw HCI commands to controller.
I think lecc command never supported proper canceling of pending connection on timeout (neither provided helper command for this).
 
Just send LE Create Connection Cancel command before restarting scan
sudo hcitool -i hci0 cmd 0x08 0x000e

And I'd recommend using bluetoothd interfaces instead of hcitool (which may interfere with kernel HCI traffic)