zephyrproject-rtos/zephyr #33693: cmake -E env: unknown option '-Wno-unique_unit_address_if_enabled'

nandojve, 2021-03-25:
**Describe the bug**
The #31960 introduced the **pre_dt_board.cmake** file in many boards. The new rules add _EXTRA_DTC_FLAGS "-Wno-unique_unit_address_if_enabled"_ to build system. This rules if not proper defined will break _make menuconfig_.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world
2. west build -b atsamr21_xpro
3. west build -t menuconfig
4. See error

**Expected behavior**
**west build -t menuconfig** should never be broken by any change in Zephyr configuration.

**Impact**
It is not possible configure Zephyr without rollback.

**Logs and console output**
```
-- west build: running target menuconfig
...
cmake -E env: unknown option '-Wno-unique_unit_address_if_enabled'
...
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK sdk-0.12.3
 - Commit #mainline

**Additional context**
It is expected a test that can check broken Kconfig options

chrta, 2021-03-25:
Is this fixed with https://github.com/zephyrproject-rtos/zephyr/pull/32753 ?

tejlmand, 2021-03-25:
> Is this fixed with #32753 ?

I believe so.

nandojve, 2021-03-25:
> > Is this fixed with #32753 ?
> 
> I believe so.

Works For Me!

chrta, 2021-04-01:
With the merge, this issue is now fixed and can be closed. I verified this with the current master and "Steps to reproduce"

galak, 2021-04-01:
Closing this, @nandojve feel free to re-open if there is still an issue.

closed 2021-04-01 by galak