zephyrproject-rtos/zephyr #33702: build error for esp32 whenSSD1306 is enabled

houssine97, 2021-03-25:
**Describe the bug**
When building a project with the SSD1306 enabled and set up, I get build errors  for example:

`../include/devicetree.h:300:40: error: 'DT_N_INST_0_solomon_ssd1306fb_P_prechargep' undeclared (first use in this function)
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                                        ^~~~~~~~~
../include/devicetree.h:2210:24: note: in definition of macro 'DT_CAT'
 #define DT_CAT(a1, a2) a1 ## a2
                        ^~
../include/devicetree.h:1726:34: note: in expansion of macro 'DT_PROP'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                  ^~~~~~~
../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
../include/devicetree.h:1704:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
../include/devicetree.h:1726:42: note: in expansion of macro 'DT_DRV_INST'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                          ^~~~~~~~~~~
C:/Users/houssine/zephyrproject/zephyr/drivers/display/ssd1306.c:120:3: note: in expansion of macro 'DT_INST_PROP'
   DT_INST_PROP(0, prechargep),
   ^~~~~~~~~~~~
../include/devicetree.h:300:40: note: each undeclared identifier is reported only once for each function it appears in
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                                        ^~~~~~~~~
../include/devicetree.h:2210:24: note: in definition of macro 'DT_CAT'
 #define DT_CAT(a1, a2) a1 ## a2
                        ^~
../include/devicetree.h:1726:34: note: in expansion of macro 'DT_PROP'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                  ^~~~~~~
../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
../include/devicetree.h:1704:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
../include/devicetree.h:1726:42: note: in expansion of macro 'DT_DRV_INST'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                          ^~~~~~~~~~~
C:/Users/houssine/zephyrproject/zephyr/drivers/display/ssd1306.c:120:3: note: in expansion of macro 'DT_INST_PROP'
   DT_INST_PROP(0, prechargep),
   ^~~~~~~~~~~~
C:/Users/houssine/zephyrproject/zephyr/drivers/display/ssd1306.c: In function 'ssd1306_set_hardware_config':
../include/devicetree.h:300:40: error: 'DT_N_INST_0_solomon_ssd1306fb_P_display_offset' undeclared (first use in this function)
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                                        ^~~~~~~~~
../include/devicetree.h:2210:24: note: in definition of macro 'DT_CAT'
 #define DT_CAT(a1, a2) a1 ## a2
                        ^~
../include/devicetree.h:1726:34: note: in expansion of macro 'DT_PROP'
 #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
                                  ^~~~~~~
../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
../include/devicetree.h:1704:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
[... 369 lines cut ...]
../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
../include/devicetree.h:1704:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
../include/device.h:231:19: note: in expansion of macro 'DT_DRV_INST'
  DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
                   ^~~~~~~~~~~
C:/Users/houssine/zephyrproject/zephyr/drivers/display/ssd1306.c:458:1: note: in expansion of macro 'DEVICE_DT_INST_DEFINE'
 DEVICE_DT_INST_DEFINE(0, ssd1306_init, device_pm_control_nop,`

**To Reproduce**
1 - add `CONFIG_SSD1306=y` to `prj.conf`
2-Build using `west build -b esp32 samples\display\cfb`

**Impact**
code doesn't build.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK

galak, 2021-03-25:
The esp32 board devicetree doesn't support an arduino header connector at this time so this is not expected to work.  This is not a bug, but would be an enhancement request (assuming the board has an arduino header).

closed 2021-03-25 by galak

moCork, 2021-03-25:
> The esp32 board devicetree doesn't support an arduino header connector at this time so this is not expected to work. This is not a bug, but would be an enhancement request (assuming the board has an arduino header).

Hi @galak what do you mean by it does not support the arduino header. We are using Oled with I2C port.
Can you please explain what you mean