zephyrproject-rtos/zephyr #33726: test:mimxrt1010_evk: kernel_threads_sched_userspace cases meet out our space

hakehuang, 2021-03-26:
**Describe the bug**
kernel_threads_sched_userspace cases meet out our space

**To Reproduce**
Steps to reproduce the behavior:
tests/kernel/sched/schedule_api
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1010_evk ..
3. make
4. See error

**Expected behavior**
in former build this cases are PASS,but now the code size is too large

**Impact**
unknown

**Logs and console output**
```
+ docker exec confident_sinoussi build_zephyr_elf.sh mimxrt1010_evk_kernel3_master tests/kernel/sched/schedule_api mimxrt1010_evk build_900f0a3 -DCONF_FILE=prj_dumb.conf -DCONFIG_TIMESLICING=n kernel.scheduler.dumb_no_timeslicing tests/kernel/sched/schedule_api -DCONF_FILE=prj_dumb.conf -DCONFIG_TIMESLICING=n

/build/src/workspace/mimxrt1010_evk_kernel3_master

Including boilerplate (Zephyr base): /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/cmake/app/boilerplate.cmake

-- Application: /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api

-- Zephyr version: 2.5.99 (/build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr)

-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 

-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")

-- Board: mimxrt1010_evk

-- Cache files will be written to: /root/.cache/zephyr

-- Using toolchain: zephyr 0.12.2 (/opt/zephyr-sdk)

-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")

-- Found BOARD.dts: /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts

-- Generated zephyr.dts: /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/zephyr.dts

-- Generated devicetree_unfixed.h: /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/include/generated/devicetree_unfixed.h

-- Generated device_extern.h: /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/include/generated/device_extern.h

Parsing /build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/Kconfig

Loaded configuration '/build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk_defconfig'

Merged configuration 'prj_dumb.conf'

Merged configuration '/build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/misc/generated/extra_kconfig_options.conf'

Configuration saved to '/build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/.config'

Kconfig header saved to '/build/src/workspace/mimxrt1010_evk_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_900f0a3/zephyr/include/generated/autoconf.h'

-- The C compiler identification is GNU 10.2.0

-- The CXX compiler identification is GNU 10.2.0

-- The ASM compiler identification is GNU

-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc

-- Configuring done
[... 485 lines cut ...]
           OCRAM:          0 GB        64 KB      0.00%

           FLASH:       94812 B        16 MB      0.57%

            ITCM:          0 GB        32 KB      0.00%

            SRAM:       39648 B        32 KB    121.00%

        IDT_LIST:          0 GB         2 KB      0.00%

collect2: error: ld returned 1 exit status

make[2]: *** [zephyr/CMakeFiles/zephyr_final.dir/build.make:146: zephyr/zephyr.elf] Error 1

make[1]: *** [CMakeFiles/Makefile2:2562: zephyr/CMakeFiles/zephyr_final.dir/all] Error 2

make: *** [Makefile:84: all] Error 2

script returned exit code 2
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-1463-gc59cf6008b60

hakehuang, 2021-03-26:
@MaureenHelm  looks like the kernel case in userspace is too large, not sure whether I need filter out those cases from rt1010?

MaureenHelm, 2021-03-26:
> looks like the kernel case in userspace is too large, not sure whether I need filter out those cases from rt1010?

I'd like to understand what caused the SRAM to grow from <32KB to 39KB before we consider that.

dleach02, 2021-05-21:
I walked this back to 2.3 and it still wouldn't fit. I think the simple fix I'll submit is to switch this platform over to using OCRAM instead of DTCM as that will give us plenty of RAM. Ultimately, it would be nice to get Zephyr to support multiple RAM blocks.

Also note that the test application has some large RAM usage because it creates 10 threads with each having their own stack defined. There are also a couple of other large chunks of RAM used.

PR #35564 (2021-05-25): boards: mimxrt1010_evk: Point SRAM to OCRAM
Point the SRAM reference to the larger OCRAM memory block.

Fixes: #33726
commit a7c732db (2021-05-25): boards: mimxrt1010_evk: Point SRAM to OCRAM

Point the SRAM reference to the larger OCRAM memory block.

Fixes: #33726
closed 2021-05-25 by nashif