zephyrproject-rtos/zephyr #33727: mec15xxevb_assy6853: multiple tests failed due to assertion failure at WEST_TOPDIR/zephyr/kernel/sched.c:841

s-kelvin, 2021-03-26:
**Describe the bug**
There are four testcases failing: tests/drivers/adc/adc_api/, tests/drivers/gpio/gpio_api_1pin/, tests/kernel/mem_slab/mslab_api/, tests/subsys/power/power_mgmt_soc/.
All show the following same assertion failure in the log
```
ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/sched.c:841
E: r0/a1:  0x00000004  r1/a2:  0x00000349  r2/a3:  0x001184f0
E: r3/a4:  0x000e1a3d r12/ip:  0xa0000000 r14/lr:  0x000e5427
E:  xpsr:  0x4100005e
E: Faulting instruction address (r15/pc): 0x000e783a
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x1184f0 (idle 00)
E: Halting system
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mec15xxevb_assy6853 -p auto <any one of the testsuite mentioned above>
e.g. west build -b mec15xxevb_assy6853 -p auto tests/drivers/adc/adc_api/
2. west flash
3. See error

**Expected behavior**
Test should pass.

**Logs and console output**
Log from tests/drivers/adc/adc_api/
```
*** Booting Zephyr OS build zephyr-v2.5.0-1523-ge1b840c4ab52  ***
Running test suite adc_basic_test
===================================================================
START - test_adc_sample_one_channel
Samples read: 0x0af6 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 
PASS - test_adc_sample_one_channel in 0.7 seconds
===================================================================
START - test_adc_sample_two_channels
Samples read: 0x0af6 0x0af3 0xffff8000 0xffff8000 0xffff8000 0xffff8000 
PASS - test_adc_sample_two_channels in 0.7 seconds
===================================================================
START - test_adc_asynchronous_call
Samples read: 0x0af6 0x0af0 0x0af0 0x0af0 0x0af0 0xffff8000 
PASS - test_adc_asynchronous_call in 0.6 seconds
===================================================================
START - test_adc_sample_with_interval
sample_with_interval_callback: sampling 0
sample_with_interval_callback: sampling 1
sample_with_interval_callback: sampling 2
sample_with_interval_callback: sampling 3
sample_with_interval_callback: sampling 4
Samples read: 0x0af0 0x0af0 0x0af6 0x0aef 0x0aee 0xffff8000 
PASS - test_adc_sample_with_interval in 0.410 seconds
===================================================================
START - test_adc_repeated_samplings
repeated_samplings_callback: done 1
Samples read: 0x0af6 0x0af3 0xffff8000 0xffff8000 0xffff8000 0xffff8000 
repeated_samplings_callback: done 2
Samples read: 0x0af6 0x0af3 0x0af6 0x0af3 0xffff8000 0xffff8000 
repeated_samplings_callback: done 3
Samples read: 0x0af6 0x0af3 0x0af6 0x0af3 0xffff8000 0xffff8000 
repeated_samplings_callback: done 4
Samples read: 0x0af6 0x0af3 0x0af0 0x0af3 0xffff8000 0xffff8000 
repeated_samplings_callback: done 5
Samples read: 0x0af6 0x0af3 0x0af6 0x0af3 0xffff8000 0xffff8000 
repeated_samplings_callback: done 6
Samples read: 0x0af6 0x0af3 0x0af6 0x0af2 0xffff8000 0xffff8000 
repeated_samplings_callback: done 7
Samples read: 0x0af6 0x0af3 0x0aee 0x0af1 0xffff8000 0xffff8000 
repeated_samplings_callback: done 8
Samples read: 0x0af6 0x0af3 0x0aef 0x0af3 0xffff8000 0xffff8000 
ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/sched.c:841
[... 184 lines cut ...]
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/sched.c:841
E: r0/a1:  0x00000004  r1/a2:  0x00000349  r2/a3:  0x001185a0
E: r3/a4:  0x000e1a2d r12/ip:  0xa0000000 r14/lr:  0x000e537f
E:  xpsr:  0x4100007f
E: Faulting instruction address (r15/pc): 0x000e6d48
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x1185a0 (idle 00)
E: Halting system
```

**Environment (please complete the following information):**
 - Linux Ubuntu 18.04.4 LTS
 - Toolchain zephyr-sdk-0.12.3
 - zephyr-v2.5.0-1523-ge1b840c4ab52

scottwcpg, 2021-03-30:
Reproduced on EVB.  Issue related to power management.  Disable PM by CONFIG_SOC_POWER_MANAGEMENT=n in the board config and the test passes.  Any power management changes recently?

albertofloyd, 2021-03-30:
@nashif @ceolin does SoC module need to be adjusted after recent changes into PM subsystem?

ceolin, 2021-03-30:
> @nashif @ceolin does SoC module need to be adjusted after recent changes into PM subsystem?

Can you test again with upstream code ? This issue seems to be fixed in https://github.com/zephyrproject-rtos/zephyr/commit/58a0ba6dbb28413bbad19d2eb41e1f3524b0e4fa

scottwcpg, 2021-03-31:
That fixed it! Test runs to completion with all PASS.

closed 2021-03-31 by scottwcpg