zephyrproject-rtos/zephyr #33753: LVGL output doesn't match the LVGL TFT simulator for gauge widget

jeremyherbert, 2021-03-28:
**Describe the bug**

I am trying to use LVGL with a small monochrome display (128x64, SH1106). If I try to draw a gauge on the screen with a line width of 1, the needle line only has a width of 1 pixel when it is near vertical or horizontal, otherwise it has a width of 2 or more pixels. Some other scale lines are smudged also.

[image]

I had assumed this was an issue with LVGL, but if I take this exact same code and run it in the LVGL example simulator here: https://github.com/lvgl/lv_sim_eclipse_sdl (at the tagged revision v7.6.1 to match the current version in zephyr) the lines are correctly drawn as 1 pixel wide under all circumstances. 

[image]

The only adjustments I made to the simulator were:
- change to 1 bit colour
- change display resolution to 128x64
- make the default theme "empty", with font `unscii_8`

This should be the same as the Zephyr build. 

I also have an older dev kit running a zephyr build from about a year ago (unsure exactly what version), and it doesn't have this problem, lines are all correctly drawn as 1px wide:

[image]

**To Reproduce**

prj.conf
```
CONFIG_GPIO=y

CONFIG_HEAP_MEM_POOL_SIZE=16384
CONFIG_MAIN_STACK_SIZE=4096

CONFIG_DEBUG_OPTIMIZATIONS=y
CONFIG_ASSERT=y

CONFIG_I2C=y

CONFIG_DISPLAY=y
CONFIG_DISPLAY_LOG_LEVEL_DBG=y
CONFIG_LOG_STRDUP_BUF_COUNT=16

CONFIG_SSD1306=y
CONFIG_SSD1306_SH1106_COMPATIBLE=y

CONFIG_LOG=y

CONFIG_LVGL=y
CONFIG_LVGL_DISPLAY_DEV_NAME="SSD1306"
CONFIG_LVGL_USE_THEME_EMPTY=y

CONFIG_LVGL_LOG_LEVEL_INF=y

CONFIG_LVGL_USE_LABEL=y
CONFIG_LVGL_USE_BAR=y
CONFIG_LVGL_USE_LINEMETER=y
CONFIG_LVGL_USE_GAUGE=y

CONFIG_LVGL_COLOR_DEPTH_1=y
CONFIG_LVGL_BITS_PER_PIXEL=1
CONFIG_LVGL_HOR_RES_MAX=128
CONFIG_LVGL_VER_RES_MAX=64
CONFIG_LVGL_DPI=130
CONFIG_LVGL_VDB_SIZE=64

# Console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=n
CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y
CONFIG_EARLY_CONSOLE=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
```

main.c
```
#include <zephyr.h>
#include <device.h>
#include <drivers/display.h>
#include <lvgl.h>
#include <stdio.h>

void draw_gauge() {
    uint32_t count = 0U;

    static lv_style_t bg;
    lv_style_init(&bg);
    lv_style_set_bg_color(&bg, LV_STATE_DEFAULT, LV_COLOR_BLACK);
    lv_style_set_bg_grad_color(&bg, LV_STATE_DEFAULT, LV_COLOR_BLACK);
    lv_obj_add_style(lv_scr_act(), LV_OBJ_PART_MAIN, &bg);

    static lv_style_t normal_style;
    lv_style_init(&normal_style);
    lv_style_copy(&normal_style, &bg);
    lv_style_set_text_color(&normal_style, LV_STATE_DEFAULT, LV_COLOR_WHITE);
    lv_style_set_line_color(&normal_style, LV_STATE_DEFAULT, LV_COLOR_WHITE);
    lv_style_set_scale_grad_color(&normal_style, LV_STATE_DEFAULT, LV_COLOR_WHITE);
    lv_style_set_scale_end_color(&normal_style, LV_STATE_DEFAULT, LV_COLOR_WHITE);
    lv_style_set_line_width(&normal_style, LV_STATE_DEFAULT, 1);
    lv_style_set_scale_end_line_width(&normal_style, LV_STATE_DEFAULT, 1);
    lv_style_set_scale_width(&normal_style, LV_STATE_DEFAULT, 4);

    static lv_color_t needle_colors[1];
    needle_colors[0] = LV_COLOR_WHITE;

    lv_obj_t * gauge2 = lv_gauge_create(lv_scr_act(), NULL);
    lv_obj_set_size(gauge2, 60, 60);
    lv_obj_align(gauge2, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_gauge_set_needle_count(gauge2, 1, &needle_colors[0]);
    lv_obj_add_style(gauge2, LV_GAUGE_PART_MAJOR, &normal_style);
    lv_obj_add_style(gauge2, LV_GAUGE_PART_NEEDLE, &normal_style);
    lv_obj_add_style(gauge2, LV_GAUGE_PART_MAIN, &normal_style);
    lv_obj_set_style_local_pad_inner(gauge2, LV_GAUGE_PART_MAIN, LV_STATE_DEFAULT, 4);
    lv_gauge_set_scale(gauge2, 220, 21, 6);

    lv_gauge_set_value(gauge2, 0, 10);

    lv_task_handler();

    while (1) {
        if ((count % 10) == 0U) {
            lv_gauge_set_value(gauge2, 0, (count/10U) % 100);
        }
        lv_task_handler();
        k_sleep(K_MSEC(10));
        ++count;
    }
}

void main(void)
{

    struct device *display_dev;

    display_dev = device_get_binding(CONFIG_LVGL_DISPLAY_DEV_NAME);

    if (display_dev == NULL) {
        printk("device not found.  Aborting test.");
        return;
    }

    display_blanking_off(display_dev);

    draw_gauge();
}
```

Steps to reproduce the behavior:
Using the code above, build for board `nrf52840dk_nrf52840`

**Expected behavior**
Behaviour should match LVGL simulator in all cases.

**Impact**
Annoyance, reduces usefulness of simulator to test changes on a PC.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 4ce908a2342494c19df09f347ce79ab81b415185

embeddedt, 2021-03-28:
I suspect this is caused by Zephyr's changes to the default LVGL settings (in particular, antialiasing may be turned off).

jeremyherbert, 2021-03-28:
I did have a play with antialiasing, both `CONFIG_LVGL_ANTIALIAS=y` and `CONFIG_LVGL_ANTIALIAS=n` don't seem to make a difference to the output.

vanwinkeljan, 2021-04-07:
@jeremyherbert Did you use the exact same config in the simulator form the description above I assume not?

Did you test with the `native_posix` board and did it behave as expected?

Further on:

> Expected behavior
> Behaviour should match LVGL simulator in all cases.

This in only valid if the exact same config is used

> Impact
> Annoyance, reduces usefulness of simulator to test changes on a PC.

I would assume that the `native_posix` board is used for this use case and not the stock LVGL simulator

jeremyherbert, 2021-04-07:
I did try to match the configs as best as possible (it is not trivial because LVGL is not configured by Kconfig options). With the default config the two look very different to eachother.

I don't know how to use the native_posix board with a display, but I will try to work it out and test it ASAP

closed 2021-06-22 by github-actions[bot]