zephyrproject-rtos/zephyr #33754: xtensa sys timer Interrupt bug？

hongshui3000, 2021-03-28:
**Describe the bug**
When my dsp runs 32M, my program will continue to interrupt and eventually the program will crash.
I think the program will update and clear the interrupt regularly in the z_clock_announce function, but my code did not reach this place, the second interrupt came again and started to execute the interrupt function.

file: xtensa_sys_timer.c
```
#define CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC 32000000
#define CONFIG_SYS_CLOCK_TICKS_PER_SEC 10000

static void ccompare_isr(const void *arg)
{
	ARG_UNUSED(arg);
	k_spinlock_key_t key = k_spin_lock(&lock);
	uint32_t curr = ccount();
	uint32_t dticks = (curr - last_count) / CYC_PER_TICK;

	last_count += dticks * CYC_PER_TICK;

	if (!IS_ENABLED(CONFIG_TICKLESS_KERNEL)) {
		uint32_t next = last_count + CYC_PER_TICK;

		if ((int32_t)(next - curr) < MIN_DELAY) {
			next += CYC_PER_TICK;
		}
		set_ccompare(next);
	}

	k_spin_unlock(&lock, key);
	z_clock_announce(IS_ENABLED(CONFIG_TICKLESS_KERNEL) ? dticks : 1);
}
```
I changed the code to the following, the system can work normally, but I am not sure whether my changes will affect other functions
```
static void ccompare_isr(const void *arg)
{
	ARG_UNUSED(arg);
	irq_disable(TIMER_IRQ);
	k_spinlock_key_t key = k_spin_lock(&lock);
	uint32_t curr = ccount();
	uint32_t dticks = (curr - last_count) / CYC_PER_TICK;

	last_count += dticks * CYC_PER_TICK;

	if (!IS_ENABLED(CONFIG_TICKLESS_KERNEL)) {
		uint32_t next = last_count + CYC_PER_TICK;

		if ((int32_t)(next - curr) < MIN_DELAY) {
			next += CYC_PER_TICK;
		}
		set_ccompare(next);
	}

	k_spin_unlock(&lock, key);
	z_clock_announce(IS_ENABLED(CONFIG_TICKLESS_KERNEL) ? dticks : 1);
	irq_enable(TIMER_IRQ);
}

```

dcpleung, 2021-03-30:
https://github.com/zephyrproject-rtos/zephyr/blob/ab5d8432d36915e53487a0613436477e864492d1/arch/xtensa/core/xtensa-asm2.c#L147 generates the interrupt handlers which eventually calls the ISRs. It clears the interrupt only after ISR returns. This code has been running on QEMU and various Xtensa platforms without any issues. As Xtensa core is highly configurable, maybe there are additional steps needed for your core? Without the same hardware, I can't reproduce this on my side to start figuring out what would be a proper fix.

nashif, 2021-05-31:
no response for 2 months, closing

closed 2021-05-31 by nashif