zephyrproject-rtos/zephyr #33784: [Coverity CID: 220452] Out-of-bounds access in subsys/net/l2/ethernet/ethernet.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/net/l2/ethernet/ethernet.c

Category: Memory - corruptions
Function: `net_eth_carrier_off`
Component: Networking
CID: [220452](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220452)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/net/l2/ethernet/ethernet.c#L1048

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2021-03-31:
False positive, not an issue with atomic_test_and_set

galak, 2021-03-31:
Marking this as a false positive.

closed 2021-03-31 by galak