zephyrproject-rtos/zephyr #33788: [Coverity CID: 220448] Out-of-bounds access in tests/subsys/modbus/src/test_modbus_raw.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/tests/subsys/modbus/src/test_modbus_raw.c

Category: Memory - corruptions
Function: `server_raw_cb`
Component: Tests
CID: [220448](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220448)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/tests/subsys/modbus/src/test_modbus_raw.c#L28

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2021-04-23:
@jfischer-no are you going to submit a PR for this?

carlescufi, 2021-05-05:
This has been fixed.

closed 2021-05-05 by carlescufi