zephyrproject-rtos/zephyr #33798: [Coverity CID: 220438] Out-of-bounds access in subsys/bluetooth/audio/vocs_client.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/bluetooth/audio/vocs_client.c

Category: Memory - corruptions
Function: `vocs_discover_func`
Component: Bluetooth
CID: [220438](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220438)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/bluetooth/audio/vocs_client.c

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

Thalley, 2021-03-29:
This error seem to occur to all occurrences of of `bt_uuid_cmp` for both VOCS (vocs.c) and VOCS client (vocs_client.c). 

To me it seems more like an issue with `bt_uuid_cmp` than the VOCS implementations.

Thalley, 2021-03-29:
This is similar to https://github.com/zephyrproject-rtos/zephyr/issues/33798#event-4521897901

joerchan, 2021-03-29:
@Thalley You linked this issue to itself, did you mean another one?

Thalley, 2021-03-29:
> @Thalley You linked this issue to itself, did you mean another one?

Yes, thanks! :)

joerchan, 2021-03-29:
> This error seem to occur to all occurrences of of bt_uuid_cmp for both VOCS (vocs.c) and VOCS client (vocs_client.c).

It does appear for almost all occurrences of `bt_uuid_cmp`, we have marked them all as false positives.
Not sure how to fix it, but coverity clearly sees that the UUID is of type 16, and then says there would be out-of-bounds access if you take the type 128 switch case.

If you know how to make coverity stop reporting these it would help a lot with reducing noise from these reports.

Thalley, 2021-03-29:
> It does appear for almost all occurrences of `bt_uuid_cmp`, we have marked them all as false positives.
> Not sure how to fix it, but coverity clearly sees that the UUID is of type 16, and then says [...]

Thanks, that was also my assumption of what has been done previously. 

Unfortunately (as mentioned on Slack) it isn't possible to run coverity locally, so I can't easily test things. I'm not sure what really causes the "which accesses it at byte offset 16." issue though.

dleach02, 2021-03-31:
guys, I think the fix for this is in bt_uuid_cmp() function. The first test is if the types don't align to just go do a uuid128_cmp() which generically copies the UUID from both parameters as if they are 128bit but the original second parameter of BT_UUID_VOCS_DESCRIPTION is a UUID16.

I think you want uuid128_cmp() to detect the size of the UUIDs being compared and return false if they are not both 128bit.

dleach02, 2021-03-31:
Nevermind. This is a false positive. Coverity is picking up on the fact that the uuid_to_uuid128() switch statement could access past the size but it isn't able to tell that the parameter passed in would have a type of 16.

Thalley, 2021-04-01:
@dleach02 Yeah, thanks, we marked it as false positive, but I forgot to close the Github issues :)

Thalley, 2021-04-01:
Marked as false positive.

closed 2021-04-01 by Thalley