zephyrproject-rtos/zephyr #33799: [Coverity CID: 220437] Wrong size argument in tests/lib/mem_alloc/src/main.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/tests/lib/mem_alloc/src/main.c

Category: Incorrect expression
Function: `test_no_mem_malloc`
Component: Tests
CID: [220437](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220437)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/tests/lib/mem_alloc/src/main.c#L153

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

enjiamai, 2021-03-30:
This is a false positive, the testing intent to give it the wrong (not align) size then the allocation will fail.  Please close it.

nashif, 2021-04-14:
closed in coverity

enjiamai, 2021-05-10:
@nashif already closed in coverity.