zephyrproject-rtos/zephyr #33808: [Coverity CID: 220428] Out-of-bounds access in subsys/bluetooth/audio/vocs.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/bluetooth/audio/vocs.c#L325

Category: Memory - corruptions
Function: `bt_vocs_init`
Component: Bluetooth
CID: [220428](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220428)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/bluetooth/audio/vocs.c

```
313          * also update the characteristic declaration (always found at [i - 1]) with the
314          * BT_GATT_CHRC_WRITE_WITHOUT_RESP property.
315          */
316         for (int i = 1; i < vocs->srv.service_p->attr_count; i++) {
317             attr = &vocs->srv.service_p->attrs[i];
318    
>>>     CID 220428:    (OVERRUN)
>>>     Overrunning array "struct bt_uuid_16 [1]({{.uuid = {BT_UUID_TYPE_16}, .val = 11137}})" of 4 bytes by passing it to a function which accesses it at byte offset 16.
319             if (init->location_writable && !bt_uuid_cmp(attr->uuid, BT_UUID_VOCS_LOCATION)) {
320                 /* Update attr and chrc to be writable */
321                 chrc = vocs->srv.service_p->attrs[i - 1].user_data;
322                 attr->write = write_location;
323                 attr->perm |= BT_GATT_PERM_WRITE_ENCRYPT;
324                 chrc->properties |= BT_GATT_CHRC_WRITE_WITHOUT_RESP;
319             if (init->location_writable && !bt_uuid_cmp(attr->uuid, BT_UUID_VOCS_LOCATION)) {
320                 /* Update attr and chrc to be writable */
321                 chrc = vocs->srv.service_p->attrs[i - 1].user_data;
322                 attr->write = write_location;
323                 attr->perm |= BT_GATT_PERM_WRITE_ENCRYPT;
324                 chrc->properties |= BT_GATT_CHRC_WRITE_WITHOUT_RESP;
>>>     CID 220428:    (OVERRUN)
>>>     Overrunning array "struct bt_uuid_16 [1]({{.uuid = {BT_UUID_TYPE_16}, .val = 11139}})" of 4 bytes by passing it to a function which accesses it at byte offset 16.
325             } else if (init->desc_writable &&
326                    !bt_uuid_cmp(attr->uuid, BT_UUID_VOCS_DESCRIPTION)) {
327                 /* Update attr and chrc to be writable */
328                 chrc = vocs->srv.service_p->attrs[i - 1].user_data;
329                 attr->write = write_output_desc;
330                 attr->perm |= BT_GATT_PERM_WRITE_ENCRYPT;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

Thalley, 2021-03-29:
This is similar to https://github.com/zephyrproject-rtos/zephyr/issues/33798#event-4521897901

Thalley, 2021-04-01:
Marked as false positive.

closed 2021-04-01 by Thalley