zephyrproject-rtos/zephyr #33812: [Coverity CID: 220424] Out-of-bounds access in drivers/watchdog/wdt_mcux_imx_wdog.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/drivers/watchdog/wdt_mcux_imx_wdog.c#L75

Category: Memory - corruptions
Function: `mcux_wdog_install_timeout`
Component: Drivers
CID: [220424](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220424)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/drivers/watchdog/wdt_mcux_imx_wdog.c#L75

```
69    
70         if (data->timeout_valid) {
71             LOG_ERR("No more timeouts can be installed");
72             return -ENOMEM;
73         }
74    
>>>     CID 220424:  Memory - corruptions  (OVERRUN)
>>>     Overrunning struct type wdog_config_t of 14 bytes by passing it to a function which accesses it at byte offset 27.
75         WDOG_GetDefaultConfig(&data->wdog_config);
76         data->wdog_config.interruptTimeValue = 0U;
77    
78         data->wdog_config.timeoutValue =
79               WDOG_TMOUT_SEC(cfg->window.max);
80    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2021-03-29:
There's no overrun here. Is Coverity confused about the size of `wdog_config_t`? Closing as false positive.

```c
void WDOG_GetDefaultConfig(wdog_config_t *config)
{
    assert(NULL != config);

    /* Initializes the configure structure to zero. */
    (void)memset(config, 0, sizeof(*config));

    config->enableWdog             = true;
    config->workMode.enableWait    = false;
    config->workMode.enableStop    = false;
    config->workMode.enableDebug   = false;
    config->enableInterrupt        = false;
    config->softwareResetExtension = false;
    config->enablePowerDown        = false;
    config->timeoutValue           = 0xffu;
    config->interruptTimeValue     = 0x04u;
    config->enableTimeOutAssert    = false;
}
```

cc: @dleach02 @hakehuang

closed 2021-03-29 by MaureenHelm

hakehuang, 2021-03-30:
maybe it is because coverity linked to another header file
```
typedef struct _wdog_config
{
    bool enableWdog;                  /*!< Enables or disables WDOG */
    wdog_clock_source_t clockSource;  /*!< Clock source select */
    wdog_clock_prescaler_t prescaler; /*!< Clock prescaler value */
    wdog_work_mode_t workMode;        /*!< Configures WDOG work mode in debug stop and wait mode */
    bool enableUpdate;                /*!< Update write-once register enable */
    bool enableInterrupt;             /*!< Enables or disables WDOG interrupt */
    bool enableWindowMode;            /*!< Enables or disables WDOG window mode */
    uint32_t windowValue;             /*!< Window value */
    uint32_t timeoutValue;            /*!< Timeout value */
} wdog_config_t;
```

dleach02, 2021-03-30:
When I saw this in the past I think it was because the coverity scans were combining different architectures when running.... there could also be an issue with different compiler options.