zephyrproject-rtos/zephyr #33819: [Coverity CID: 220417] Out-of-bounds access in subsys/modbus/modbus_core.c

zephyrbot, 2021-03-29:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/modbus/modbus_core.c#L206

Category: Memory - corruptions
Function: `modbus_init_iface`
Component: Other
CID: [220417](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220417)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/169144afa1826511ee6ec3f53d590b2c0d39d3d4/subsys/modbus/modbus_core.c#L206

```
200             LOG_ERR("Interface %u not available", iface);
201             return NULL;
202         }
203    
204         ctx = &mb_ctx_tbl[iface];
205    
>>>     CID 220417:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Passing "&ctx->state" to function "atomic_test_and_set_bit" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
206         if (atomic_test_and_set_bit(&ctx->state, MODBUS_STATE_CONFIGURED)) {
207             LOG_ERR("Interface allready used");
208             return NULL;
209         }
210    
211         k_mutex_init(&ctx->iface_lock);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2021-03-30:
there are multiple coverity errors related to atomic_test_and_set_bit(). Not sure what changed or what the problem is?

galak, 2021-03-31:
Marking it as a false positive.

closed 2021-03-31 by galak