zephyrproject-rtos/zephyr #33843: ESP32 example the log gets stuck

tiborrr, 2021-03-29:
**Describe the bug**
When flashing the ESP32 example the log gets stuck and only returns `<err> net_arp: Gateway not set for iface 0x3ffb0110` on ESP32-WROOM-32

**To Reproduce**
in the samples/board/esp32/wifi_station example
Configure SSID and password
Run
`west build -p always -b esp32`
Run and check log
`west flash;minicom`

**Expected behavior**
ESP32 shoud connect to router. Below you see the ESP32 succeeding to connect, which sometimes happens, but most of the time it fails to connect.

```inf> esp_event: WIFI_EVENT_STA_START                                                                                                                                   
I (2239) wifi:new:<11,2>, old:<1,1>, ap:<255,255>, sta:<11,2>, prof:1                                                                                                                      
I (2887) wifi:state: init -> auth (b0)                                                                                                                                                     
I (2894) wifi:state: auth -> assoc (0)                                                                                                                                                     
I (2902) wifi:state: assoc -> run (10)                                                                                                                                                     
I (3001) wifi:connected with Router 66, aid = 5, channel 11, 40D, bssid = aa:bb:cc:dd:ee:ff                                                                                                
I (3002) wifi:security: WPA2-PSK, phy: bgn, rssi: -77                                                                                                                                      
I (3003) wifi:pm start, type: 1                                                                                                                                                            
                                                                                                                                                                                           
I (3040) wifi:AP's beacon interval = 102400 us, DTIM period = 1                                                                                                                            
[00:00:03.007,000] <inf> esp_event: WIFI_EVENT_STA_CONNECTED                                                                                                                               
[00:00:05.910,000] <err> net_arp: Gateway not set for iface 0x3ffb0110                                                                                                                     
[00:00:08.879,000] <err> net_arp: Gateway not set for iface 0x3ffb0110                                                                                                                     
[00:00:09.392,000] <err> net_arp: Gateway not set for iface 0x3ffb0110                                                                                                                     
[00:00:11.952,000] <err> net_arp: Gateway not set for iface 0x3ffb0110                                                                                                                     
[00:00:17.380,000] <inf> net_dhcpv4: Received: 192.168.1.217                                                                                                                               
[00:00:17.380,000] <inf> esp32_wifi_sta: Your address: 192.168.1.217                                                                                                                       
[00:00:17.380,000] <inf> esp32_wifi_sta: Lease time: 86400 seconds                                                                                                                         
[00:00:17.380,000] <inf> esp32_wifi_sta: Subnet: 255.255.255.0                                                                                                                             
[00:00:17.380,000] <inf> esp32_wifi_sta: Router: 192.168.1.1```

**Impact**
showstopper

**Logs and console output**
```��I (724) wifi:wifi driver task: 3ffb1ef0, prio:2, stack:3584, core=0                                                                                                                      
I (725) wifi:wifi firmware version: 1865b55                                                                                                                                                
I (725) wifi:wifi certification version: v7.0                                                                                                                                              
I (727) wifi:config NVS flash: disabled                                                                                                                                                    
I (730) wifi:config nano formating: disabled                                                                                                                                               
I (734) wifi:Init data frame dynamic rx buffer num: 32                                                                                                                                     
I (739) wifi:Init management frame dynamic rx buffer num: 32                                                                                                                               
I (744) wifi:Init management short buffer num: 32                                                                                                                                          
I (749) wifi:Init dynamic tx buffer num: 32                                                                                                                                                
I (753) wifi:Init static rx buffer size: 1600                                                                                                                                              
I (757) wifi:Init static rx buffer num: 10                                                                                                                                                 
I (761) wifi:Init dynamic rx buffer num: 32                                                                                                                                                
phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2                                                                                                                                    
I (890) wifi:mode : softAP (3c:61:05:30:01:89)                                                                                                                                             
I (891) wifi:Total power save buffer number: 16                                                                                                                                            
I (891) wifi:Init max length of beacon: 752/752                                                                                                                                            
I (896) wifi:Init max length of beacon: 752/752                                                                                                                                            
*** Booting Zephyr OS build zephyr-v2.5.0-1540-g63ca48a7a9b8  ***                                                                                                                          
I (906) wifi:mode : sta (3c:61:05:30:01:88)                                                                                                                                                
[00:00:00.911,000] <inf> esp_event: WIFI_EVENT_STA_START                                                                                                                                   
I (2238) wifi:new:<11,2>, old:<1,1>, ap:<255,255>, sta:<11,2>, prof:1                                                                                                                      
I (2885) wifi:state: init -> auth (b0)                                                                                                                                                     
I (2890) wifi:state: auth -> assoc (0)                                                                                                                                                     
[... cut ...]
...                                                                                                                   
[00:03:42.330,000] <err> net_arp: Gateway not set for iface 0x3ffb0110```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 20.04
 - Toolchain: espressif (esp-2020r3-8.4.0/xtensa-esp32-elf)
 - Commit: Zephyr version: 2.5.99

**Additional context**
I could not find any info on the error "gatway not set for iface", but I believe this error get trown by `iface = net_if_get_default();`

sylvioalves, 2021-03-30:
@tiborrr Hi, would you share your project conf file? I see `CONFIG_NET_ARP=y` is used.I'll try reproducing the same configuration in here. Thanks.

tiborrr, 2021-03-30:
```
CONFIG_WIFI=y
CONFIG_WIFI_ESP32=y
CONFIG_HEAP_MEM_POOL_SIZE=98304

CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_NET_LOG=y
CONFIG_NET_SHELL=n

CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_ECP_ALL_ENABLED=y

CONFIG_ESP32_WIFI_SSID="Router 66"
CONFIG_ESP32_WIFI_PASSWORD="password"
```

even without the CONFIG_NET_ARP the gateway error appears
```
I (724) wifi:wifi driver task: 3ffb1ef0, prio:2, stack:3584, core=0                                                                                                                       
I (725) wifi:wifi firmware version: 1865b55                                                                                                                                                
I (725) wifi:wifi certification version: v7.0                                                                                                                                              
I (727) wifi:config NVS flash: disabled                                                                                                                                                    
I (731) wifi:config nano formating: disabled                                                                                                                                               
I (735) wifi:Init data frame dynamic rx buffer num: 32                                                                                                                                     
I (739) wifi:Init management frame dynamic rx buffer num: 32                                                                                                                               
I (745) wifi:Init management short buffer num: 32                                                                                                                                          
I (749) wifi:Init dynamic tx buffer num: 32                                                                                                                                                
I (753) wifi:Init static rx buffer size: 1600                                                                                                                                              
I (757) wifi:Init static rx buffer num: 10                                                                                                                                                 
I (761) wifi:Init dynamic rx buffer num: 32                                                                                                                                                
phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2                                                                                                                                    
I (892) wifi:mode : softAP (3c:61:05:30:01:89)                                                                                                                                             
I (893) wifi:Total power save buffer number: 16                                                                                                                                            
I (893) wifi:Init max length of beacon: 752/752                                                                                                                                            
I (898) wifi:Init max length of beacon: 752/752                                                                                                                                            
*** Booting Zephyr OS build zephyr-v2.5.0-1540-g63ca48a7a9b8  ***                                                                                                                          
I (908) wifi:mode : sta (3c:61:05:30:01:88)                                                                                                                                                
[00:00:00.913,000] <inf> esp_event: WIFI_EVENT_STA_START                                                                                                                                   
I (2240) wifi:new:<11,2>, old:<1,1>, ap:<255,255>, sta:<11,2>, prof:1                                                                                                                      
I (2887) wifi:state: init -> auth (b0)                                                                                                                                                     
I (2900) wifi:state: auth -> assoc (0)                                                                                                                                                     
I (2907) wifi:state: assoc -> run (10)                                                                                                                                                     
I (3010) wifi:connected with Router 66, aid = 1, channel 11, 40D, bssid = aa:bb:cc:dd:ee:ff                                                                                                
I (3010) wifi:security: WPA2-PSK, phy: bgn, rssi: -73                                                                                                                                      
[... cut ...]
[00:00:04.294,000] <inf> esp32_wifi_sta: Your address: 192.168.1.217                                                                                                                       
[00:00:04.294,000] <inf> esp32_wifi_sta: Lease time: 86400 seconds                                                                                                                         
[00:00:04.294,000] <inf> esp32_wifi_sta: Subnet: 255.255.255.0                                                                                                                             
[00:00:04.294,000] <inf> esp32_wifi_sta: Router: 192.168.1.1 
```

sylvioalves, 2021-03-30:
@tiborrr, `CONFIG_NET_ARP=y` is enabled by default with `CONFIG_NET_IPV4=y`.
The `<err> net_arp: Gateway not set for iface 0x3ffb0110` is a momentary gap between interface up event and DHCP IP assign. Once IP gets assigned, the error log ends. We will check this.

However, the log error isn't the root cause for connection issues.
You get the message `<inf> esp_event: WIFI_EVENT_STA_CONNECTED` but there is no callback regarding IP assignment?

marinbudimir, 2021-03-30:
I was getting the same error but it was because of my network's dhcp configuration, so I recommend testing on another network.

tiborrr, 2021-03-30:
> I was getting the same error but it was because of my network's dhcp configuration, so I recommend testing on another network.

OK but this might also happen when you deploy to a customer. What settings should you adjust at the router side to fix this. Furthermore, why do other devices like my computer connect and the esp32 not?

Did you figure that out @marinbudimir ?

sylvioalves, 2021-03-30:
@tiborrr I guess I was able to reproduce this after a while. I'll PR soon.

PR #35715 (2021-05-27): wifi: driver: esp32: fix reconnection issue
Device won't reconnect automatically even if
AP station is available. This fix adds the carrier event, indicating
that network is present again enabling DHCP bound event.
Also, internal wifi event callback  was added into wifi driver to enable proper
event handling.

Fixes #33843
commit 95b916d1 (2021-05-27): drivers: wifi: esp32: fix reconnect issue

Device won't reconnect automatically even if
AP station is available. This fix adds the carrier event, indicating
that network is present again enabling DHCP bound event.
Also, internal wifi event callback was added into wifi
driver to enable proper event handling.

Update west.yml to bring exposed wifi event callback.

Fixes #33843
closed 2021-05-27 by galak